ros-controls/ros2_control #2068: control_toolbox builid error

MrstupidJ, 2025-02-24:
**Describe the bug**
Hi, 

I am playing with this package (ros2_control) with ROS2 humble, but it fails when I tried to follow the [getting started doc](https://control.ros.org/humble/doc/getting_started/getting_started.html#building-from-source) to build the code. The error message came from control_toolbox package. 

**To Reproduce**
Steps to reproduce the behavior:
1. Everything looks pretty normal (we can get "#All required rosdeps installed successfully" after installing all the dependencies). 
2. The error occurs when I build the source code with: 
`colcon build --symlink-install` 
4. See error`ros2_control_ws/src/ros-controls/control_toolbox/include/control_filters/exponential_filter.hpp:25:10: fatal error: control_toolbox/exponential_filter_parameters.hpp: No such file or directory
   25 | #include "control_toolbox/exponential_filter_parameters.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

**Expected behavior**
The code should build here .

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: [e.g. ubuntu 22.04]
 - Version [e.g. Humble]

christophfroehlich, 2025-02-24:
you either have to use the humble branch of control_toolbox, or the master branch of realtime_tools. See the two different repos files linked in the getting started page, the versions have to match.

christophfroehlich, 2025-02-24:
See the repos file in the two tabs "Stable version" and "Development version".

Mehul0x, 2025-03-30:
Hi, I was looking into this bug if I could potentially solve it. Cmake is generating 

I didn't find any difference between the master and humble branches of control_toolbox. Can you point me towards the right direction? What should I look for? 

From my understanding cmake is failing to do this 
```
generate_parameter_library(
exponential_filter_parameters
src/control_filters/exponential_filter_parameters.yaml
)
```

christophfroehlich, 2025-03-30:
@Mehul0x I think the original issue was fixed by the author, as there is no new activity.
If you have further questions, please post in the [correct repository](https://github.com/ros-controls/control_toolbox/)

closed 2025-03-30 by christophfroehlich