ros-controls/ros2_control #2072: Clarification on Header-Only Implementation

my-rice, 2025-02-25:
Hello. I have noticed that features such as [semantic component interface](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/include/semantic_components/semantic_component_interface.hpp), [semantic component command interface](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/include/semantic_components/semantic_component_command_interface.hpp) and their implementations (e.g., [force torque sensor](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/include/semantic_components/force_torque_sensor.hpp), [pose sensor](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/include/semantic_components/pose_sensor.hpp), and so on) are written as header-only libraries. In contrast, other libraries, such as the [controller interface](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/src/controller_interface.cpp), [controller interface base](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/src/controller_interface_base.cpp), and [chainable controller interface](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/src/chainable_controller_interface.cpp), are structured with both source and header files. I was wondering if there is a specific reason for this difference in implementation style.
Could someone kindly explain the rationale behind this distinction? Any insights or best practices regarding the decision to use header-only libraries for certain components would be greatly appreciated.

PS: I originally asked this question in the discussion of [this PR](https://github.com/ros-controls/ros2_control/pull/370#issuecomment-2682847722), but I was advised to open an issue instead, so I am posting it here.

bmagyar, 2025-03-14:
It has to do with the difference in the nature of the two types, semantic components and controller_interface. 

The `controller_interface` classes are base classes that are meant to be *inherited from* along with some default implementation you *normally* don't care about. It's an interface class with *some* defaults. (This applies to all the ones you mentioned.) Generally speaking for such classes I prefer to *hide* the implementation, aka move it to a different file and let people focus on the interface declarations in the header file.

The semantic component classes don't really bring new functionality, they are helper classes that capture the *semantics* of a bunch of numbers in the context of a common sensor type. The files you referenced contain *templated* classes which don't form compilation units on their own, they need to be used somewhere to instantiate a class once the type parameter was supplied.

my-rice, 2025-03-14:
Thanks for the clear explanation and your time. I now understand the reason for the difference in implementation style.

closed 2025-03-14 by my-rice