ros-controls/ros2_control #2075: is it possible to get the state of a hardware_components inside a controller?

xibeisiber, 2025-02-27:
Hi, I write a controller and a hardware component. 

I wish that the controller can be activated only when the hardware component is active. Therefore I want to check the state of the hardware component in the on_activate() of the controller.

Is there any API for this?
Thanks!

xibeisiber, 2025-02-27:
I tried to call service "/controller_manager/list_hardware_components" in the on_activate() of the controller. It failed to get a response(is nullptr)

christophfroehlich, 2025-12-07:
No, there is no API for doing that. Instead of using ROS service calls I'd recommend using state interfaces for that. Either you have specific state interfaces of your hardware component already, or you add a dummy one. The ros2_controller activation fails if this interface does not exist then.