ros-controls/ros2_control #2079: Hardware interface lifecycle behavior does not follow ros2 design document

Martin-Oehler, 2025-02-28:
**Describe the bug**
The hardware interface class `SystemInterface` implements the `LifecycleNodeInterface`, but its behavior does not follow the [ros2 design document for managed nodes](https://design.ros2.org/articles/node_lifecycle.html).

As described in the document, raising an error in the primary states should lead to a transition to the `ErrorProcessing` state. Instead, the controller manager crashes with an exception. This prevents any error handling in these situations and somewhat defeats the purpose of the lifecycle logic in my opinion. This even takes down any other hardware interfaces running in the same manager, which is fatal.

Furthermore, a custom logic is implemented for state transitions with two services:
```
/controller_manager/set_hardware_component_state  #initiate state transition
/controller_manager/list_hardware_components  #lifecycle state as a field
```
Instead of the interfaces usually offered by lifecycle nodes. This prevents re-using lifecycle management logic and code. There is also no latched/transient local topic for transitions which makes external management of the lifecycle challenging.

**To Reproduce**
Return `CallbackReturn::ERROR` in either `on_configure()` or `on_activate()` (others not tested)
The controller manager node will crash with:
```
terminate called after throwing an instance of 'std::runtime_error'
what():  Failed to set the initial state of the component : hardware_interface to active
```

**Expected behavior**
The hardware interface should obey the design document and transition to the `ErrorProcessing` state.

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version jazzy

saikishor, 2025-02-28:
Hello @Martin-Oehler!

Regarding the issue, at the startup of the ros2_control this is indeed the desired behaviour. Once the startup is properly done, if any of the methods return the ERROR, the `error` transition callback is executed. 

When I use this code in production, if the robot is not able to start properly and setup all the controllers. I want it to crash and fail instead of continuing. For your case, you can control the initial state by setting the parameter `hardware_components_initial_state` (You can find more information here: https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#parameters)

I hope this answers your questions:)

bmagyar, 2025-03-02:
Also one thing is pretty certain that you should never expect to have lifecycle node services for anything ros2_control. It massively violates the framework design. We are using the API and try aligning with the state machine as close as possible so there is something familiar for people.

christophfroehlich, 2025-03-02:
I agree in terms of services and CLI, but should CM really throw an exception if something is wrong in `on_configure()` or `on_activate()`?

bmagyar, 2025-03-02:
Ah that it shouldn't. We just haven't gotten to make that bulletproof
enough.

saikishor, 2025-03-02:
I believe this was intentional only at the startup, this was introduced in https://github.com/ros-controls/ros2_control/pull/1729.

https://github.com/ros-controls/ros2_control/blob/3aa38b02283aa6bd04e288919de2724871f599af/controller_manager/src/controller_manager.cpp#L552-L564

If the user's intent is to set an initial state to the hardware component and if it is not able to set, it is better to fail initially than later

saikishor, 2025-03-02:
Post startup, any change in component state between the lifecycle states is bulletproofed already

Martin-Oehler, 2025-03-03:
I can not follow the reasoning why it should be better to crash the whole process instead of going back into `Unconfigured` as described in the design document.

Consider the following two use-cases:
* The hardware is not ready for connection yet (booting up, ..). The driver should try to repeatedly connect until successful.
* An optional hardware component fails to connect due to hardware failure.

Both these cases can not be covered by the current implementation because the whole process just crashes and takes all interfaces with it. From my understanding, the whole point of the life-cycle system is to increase robustness and prevent exactly this case.

christophfroehlich, 2025-03-04:
This is something worth to discuss. 

The point of view from @saikishor is that if someone gives the initial state (explicitly, or implicitly) of the components in the configuration files and the launch fails, then this is an unintended behavior and should to blow up.
If you know at startup that something might fail/not be available, then better to configure the components to stay unconfigured or don't load it at all and use the [hardware_spawner](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#hardware-spawner) and/or service calls later.

Martin-Oehler, 2025-03-05:
> If you know at startup that something might fail/not be available, then better to configure the components to stay unconfigured or don't load it at all and use the [...]

I do not think this would be a solution for these two examples. Even if you delay the startup process by initiating the state transition at a later point with a service call, the setup might still fail and crash unless you know with absolute certainty that the hardware is already available, leading to duplicate logic.

> because all your startup controller spawners will no more work as they cannot get their interfaces as the hardware is not in expected state.

Sorry, I do not have a deep knowledge about the controller startup process, but shouldn't controllers just wait until the respective hardware interfaces become active? I assume there is already some logic implemented for that as the configuration time of a hardware interface is unknown.

> For instance, for ethercat based systems, sometimes you have to restart the EtherCAT master, if something is not working. If my application crashes, based on the exit code I can decide to restart my EtherCAT master or not and restart my ros2_control_node app.

In some situations, the correct error handling behavior might be crashing. But in others, it may not. My point is, that the current implementation decides for you that the error will be handled with crashing.

If the `ErrorProcessing` state were to be entered instead, developers could implement their own fault recovery. And that could be throwing an exception, but also something else.

saikishor, 2025-03-05:
> I do not think this would be a solution for these two examples. Even if you delay the startup process by initiating the state transition at a later point with a service call, the setup might still [...]

If you do it at later point via service call, it won't be crashing anymore. The crash is only when the intial_state is not respected. From later on via service call, it is respecting the Lifecycle design. 

> Sorry, I do not have a deep knowledge about the controller startup process, but shouldn't controllers just wait until the respective hardware interfaces become active? I assume there is already some logic implemented for that as the configuration time of a hardware interface is unknown.

Not really, they do not wait for the hardware interface to be available. If the controller requests an interface that doesn't exist yet, then it doesn't activate and it aborts. There might be a case, where the controller requests an invalid one and there is no way we can keep it waiting expecting someone to expose that particular interface. Do you get my point?

> In some situations, the correct error-handling behavior might be crashing. But in others, it may not. My point is, that the current implementation decides for you that the error will be handled with crashing.

As long as you don't set the `hardware_components_intiial_state` parameter (https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#parameters), all are activated by default. If you have an hardware that is not deterministic in its state, I recommend you to set that particular component to unconfigured and later activate it later via services.

Martin-Oehler, 2025-03-05:
I see, thanks. I will try to implement my desired behavior with your suggestions and update this ticket with my results.

Martin-Oehler, 2025-03-05:
So I found a pretty simple solution to my problem.

As suggested, I do not initialize the hardware interface on startup by setting the parameter `hardware_components_initial_state.unconfigured' for my interface in the controller manager. This ensures that the controller manager does not crash when `on_configure()` fails.

Next, I start spawners for the hardware interface as well as for all mandatory controllers with respawn:

```
launch:
- node:
    pkg: "controller_manager"
    exec: "spawner"
    name: "joint_state_broadcaster_spawner"
    output: "screen"
    args: "joint_state_broadcaster"
    respawn: "true"
    respawn_delay: 5.0

- node:
    pkg: "controller_manager"
    exec: "hardware_spawner"
    name: "hardware_interface_spawner"
    output: "screen"
    args: "flipper_interface --activate"
    respawn: "true"
    respawn_delay: 5.0
```

This will ensure that hardware and controllers are always transitioned to the active state when possible. 

However, I still find it very unintuitive that the lifecycle transitions are different depending on whether the state is progressed by the controller manager or services interface. But as the latter works as expected, this solution works fine for me now.

christophfroehlich, 2025-03-06:
Independent of the crash-or-no-crash question, your solution here should be added to our docs. Do you have suggestions where to put that?
https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#hardware-spawner
or https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#parameters
or https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/hardware_components_userdoc.html

Martin-Oehler, 2025-03-24:
I think this page seems the most related: https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/hardware_components_userdoc.html

It also does contain information regarding the different handling of errors (transition by controller manager vs service interface). I suggest to rework and extend the last paragraph "Handling of errors that happen during read() and write() calls" to a more general "Error handling" paragraph.

Regarding my solution, I noticed some minor nit-pick. Respawning the hardware spawner repeatedly to transition the hardware interface to active works still works when the interface is already active. However, the controller spawner complains in these cases because the controller is already active and exits with an error. This error spam can make it hard to spot actual errors.

saikishor, 2025-03-24:
> It also does not contain information regarding the different handling of errors (transition by controller manager vs service interface). I suggest to rework and extend the last paragraph "Handling of errors that happen during read() and write() calls" to a more general "Error handling" paragraph.

If you can open a PR, we will be happy to review and merge it 

> Regarding my solution, I noticed some minor nit-pick. Respawning the hardware spawner repeatedly to transition the hardware interface to active works still works when the interface is already [...]

I'm not sure if I understand your case. Do you mind explaining it again?. What do you mean when the interface is already active? I don't understand the fact how can the controller can activate, if the HW is not yet available

Martin-Oehler, 2025-03-24:
> If you can open a PR, we will be happy to review and merge it

Unfortunately, I will change employment soon and will loose access to my ROS2 system. I will see what I can do.

> I'm not sure if I understand your case. Do you mind explaining it again?. What do you mean when the interface is already active? I don't understand the fact how can the controller can activate, if the HW is not yet available

For my solution, I want to ensure that A) all hardware interfaces are always active and B) that the necessary controllers (e.g. the joint state broadcaster) is always active. B) is necessary because controllers are disabled when the hardware interface registers an error. This ensures a robust behavior in case of any errors can that be recovered from.

I achieve this by starting two spawners with "respawn" and "respawn_delay" set. These will periodically attempt to switch the hardware interface as well as the controllers to the active state. Naturally, because this happens periodically and unconditionally, hardware interface and controllers will already be in the active state unless an error occurred in the meantime.

If this is the case, the hardware spawner will print out a success message on the console because the interface is already in the desired active state. The controller spawner, on the other hand, will print out an error because it was not able to switch the controller from "active" to "active". As this happens periodically, the console will be cluttered with (expected) error messages.

I hope this explanation makes it clear.

saikishor, 2025-03-27:
> These will periodically attempt to switch the hardware interface as well as the controllers to the active state. Naturally, because this happens periodically and unconditionally, hardware interface and controllers will already be in the active state unless an error occurred in the meantime.

I don't understand why you have to periodically keep calling the hardware interface and controllers to be active, just so you recover from error state and go back to active state?

> If this is the case, the hardware spawner will print out a success message on the console because the interface is already in the desired active state. The controller spawner, on the other hand, [...]

This behavior might have been fixed with https://github.com/ros-controls/ros2_control/pull/2088. Can you try it out and see if it works for you :)

tony-p, 2025-05-28:
> seem to contradict the UML Diagram in the [documentation](https://control.ros.org/jazzy/doc/getting_started/getting_started.html#resource-manager). I fully except all interfaces, and everything calling them to follow the design of life cycle nodes. What am I missing? Is the yellow box wrong? Should it be "Controllers and hardware interfaces are 'sort of managed' ROS2 nodes [...] activate, etc. Except error handling"

The detail is that the `hardware_interface`s are not nodes, they merely implement the `LifecycleNodeInterface` (not the node itself)

I remember seeing a discussion in the past over ros2_control looking at using components and a custom component manager. In this way the requirements that everything runs in one process would be met, and the hardware interfaces could be fully functional (lifecycle) nodes with parameters and all, however I don't know what the current thinking behind that is

Martin-Oehler, 2025-06-01:
> I don't understand why you have to periodically keep calling the hardware interface and controllers to be active, just so you recover from error state and go back to active state?

Yes, the idea is to always get back to a working state if the error is non-fatal.

> This behavior might have been fixed with [#2088](https://github.com/ros-controls/ros2_control/pull/2088). Can you try it out and see if it works for you :)

Thanks for the pointer. Unfortunately, I have since changed employment and can not test it anymore.

closed 2025-09-01 by github-actions[bot]