ros-controls/ros2_control #2089: `rw_rate` with multiple hardware components leads to skipped cycles

RobertWilbrandt, 2025-03-03:
**Describe the bug**
We currently have a robot system with two hardware components (a UR16e and a kuka KR50), one running at 500Hz and one running at 250Hz. The controller manager `update_rate` is set to 500 and the slower interface uses the `rw_rate` attribute:
```
<ros2_control name="${name}" type="system" rw_rate="250">
  [...]
</ros2_control>
```

With this setup, the slower hardware is not able to move smoothly and "stutters". Using the tracing in [kuka_rsi_driver](https://github.com/fzi-forschungszentrum-informatik/kuka_rsi_driver), we can clearly see that the read- and write-functions frequently "miss a step" (see screenshots), i.e. only get called after 3 cycles instead of every 2.

**Expected behavior**
The `read()` and `write()` functions of the slower hardware interface should get called in every other control cycle consistently.

**Screenshots**

Through tracing we obtained the periods between successive `read()` and `write()` calls in our hardware interface. We expect these functions to be called every ~4ms, but we can clearly see that the period often times jumps to ~6ms:
[image]

In detail:
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04.1 LTS
- Version: Jazzy
- ros2_control is built from sources

**Additional context**
It seems like the [calculation in resource_manager](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/src/resource_manager.cpp#L1788) is the cause of the problem. It currently only ensures that the read/write functions are not called too frequently by checking
`actual_period.seconds() * read_rate >= 0.99`

I suspect that jitter in the faster hardware interface execution time leads to this deciding to skip a cycle frequently. I think a better approach would be to also consider the "timing error" resulting from skipping the call in a given cycle:
`(actual_period.seconds() + 1. / resource_storage_->cm_update_rate_) * read_rate`

I am going to check if this fixes the problem for our setup and try creating a PR if successful.

saikishor, 2025-03-03:
@RobertWilbrandt Do you mind giving a shot the PR #2086 and share the diagnostics of your HW components?

Thank you

RobertWilbrandt, 2025-03-03:
Sure, i could not yet run this again on real hardware but the problem persists with two mock components. Here are the diagnostics i obtained:

[image]

```
header:
  stamp:
    sec: 1741023996
    nanosec: 514931812
  frame_id: ''
status:
- level: "\0"
  name: 'controller_manager: Controllers Activity'
  message: Not all controllers are active
  hardware_id: ros2_control
  values:
  - key: schunk_ppd_controller.state
    value: inactive
  - key: joint_trajectory_controller.state
    value: active
  - key: joint_trajectory_controller.execution_time
    value: 'Avg: 7.79 [2.13 - 799.52] us, StdDev: 6.21'
  - key: kuka_status_broadcaster.state
    value: active
  - key: kuka_status_broadcaster.execution_time
    value: 'Avg: 2.13 [0.73 - 826.37] us, StdDev: 3.27'
  - key: kuka_pose_broadcaster.state
    value: active
  - key: kuka_pose_broadcaster.execution_time
    value: 'Avg: 2.66 [0.92 - 117.01] us, StdDev: 2.32'
  - key: kuka_joint_trajectory_controller.state
    value: inactive
  - key: joint_state_broadcaster.state
    value: active
  - key: joint_state_broadcaster.execution_time
    value: 'Avg: 42.85 [13.24 - 591.20] us, StdDev: 24.28'
  - key: ur_joint_trajectory_controller.state
    value: inactive
- level: "\x02"
  name: 'controller_manager: Hardware Components Activity'
  message: '

    High periodicity jitter or mean error : [ kuka_kr50 ]'
  hardware_id: ros2_control
  values:
  - key: ur.state
    value: active
  - key: ur.read_cycle.execution_time
    value: 'Avg: 12.12 [2.04 - 629.76] us, StdDev: 13.27'
  - key: ur.write_cycle.execution_time
    value: 'Avg: 17.22 [0.02 - 217.41] us, StdDev: 13.32'
  - key: kuka_kr50.state
    value: active
  - key: kuka_kr50.read_cycle.execution_time
    value: 'Avg: 2.25 [0.26 - 57.75] us, StdDev: 1.87'
  - key: kuka_kr50.read_cycle.periodicity
    value: 'Avg: 249.40 [157.75 - 252.52] Hz, StdDev: 6.82 -> Desired : 250 Hz'
  - key: kuka_kr50.write_cycle.execution_time
    value: 'Avg: 3.64 [0.54 - 94.76] us, StdDev: 3.84'
  - key: kuka_kr50.write_cycle.periodicity
    value: 'Avg: 230.24 [148.74 - 252.53] Hz, StdDev: 34.11 -> Desired : 250 Hz'
- level: "\x02"
  name: 'controller_manager: Controller Manager Activity'
  message: 'Controller Manager has bad periodicity : 500.302890 Hz. Expected consistent 500 Hz'
  hardware_id: ros2_control
  values:
  - key: update_rate
    value: '500'
  - key: periodicity.average
    value: '500.302890'
  - key: periodicity.standard_deviation
    value: '86.005452'
  - key: periodicity.min
    value: '233.325128'
  - key: periodicity.max
    value: '34331.227685'
```

I guess the controller manager periodicity stems from me running this locally without RT or low-latency kernel, but the `kuka_kr50.[read|write]_cycle.periodicity ` seem to show the same problem i am observing: One cycle missed => 6ms period => 166Hz frequency recorded.

saikishor, 2025-03-03:
Ok it seems like it is missing for write cycles strangely, however, read cycles are very good with the same logic. If a controller or a component takes longer, it might affect the trigger rate.

RobertWilbrandt, 2025-03-03:
I would expect that, as the control loop is always `read()`->`update()`->`write()`->`sleep()` the jitter of all components will add up and make a cycle skip in write more likely. Apparently it does not take too much jitter to fall under the 0.99 * frequency cutoff.

This does however affect both functions. In the ~500s run i did to look at the diagnostics, writes had a period of more than 5ms in 20.5% of calls but reads also had that problem in 0.74% (gathered from the trace). This is the reason why `kuka_kr50.read_cycle.periodicity` also has a minimum frequency of 157.75 even if the average and standard deviation are quite good.

saikishor, 2025-03-03:
@RobertWilbrandt I set the 0.99 accounting for jitter alone, may be changing this to 0.98 helps?? Or something around that range? Can you please check?

saikishor, 2025-03-03:
> @RobertWilbrandt I set the 0.99 accounting for jitter alone, may be changing this to 0.98 helps?? Or something around that range? Can you please check?

@RobertWilbrandt if this doesn't help, I have another idea that will surely work. Please check the above and let me know please :)

RobertWilbrandt, 2025-03-03:
I did some tests with different factors.

With 0.98:
```
  - key: kuka_kr50.read_cycle.execution_time
    value: 'Avg: 2.90 [0.34 - 40.94] us, StdDev: 2.45'
  - key: kuka_kr50.read_cycle.periodicity
    value: 'Avg: 249.56 [165.45 - 255.09] Hz, StdDev: 5.83 -> Desired : 250 Hz'
  - key: kuka_kr50.write_cycle.execution_time
    value: 'Avg: 3.09 [0.55 - 50.57] us, StdDev: 3.48'
  - key: kuka_kr50.write_cycle.periodicity
    value: 'Avg: 242.87 [145.84 - 255.10] Hz, StdDev: 21.99 -> Desired : 250 Hz'

Read skips: 250/46272 => 0.54%
Write skips: 4046/46272 => 8.74%
Duration: 185.58s
```

With 0.97:
```
  - key: kuka_kr50.read_cycle.execution_time
    value: 'Avg: 2.59 [0.40 - 36.78] us, StdDev: 2.23'
  - key: kuka_kr50.read_cycle.periodicity
    value: 'Avg: 249.96 [170.18 - 257.63] Hz, StdDev: 2.14 -> Desired : 250 Hz'
  - key: kuka_kr50.write_cycle.execution_time
    value: 'Avg: 3.62 [0.63 - 90.15] us, StdDev: 3.84'
  - key: kuka_kr50.write_cycle.periodicity
    value: 'Avg: 246.50 [164.58 - 257.73] Hz, StdDev: 15.97 -> Desired : 250 Hz'

Read skips: 50/46993 => 0.11%
Write skips: 2358/46993 => 5.02%
Duration: 188.07s
```

This shows that a lower factor actually makes the problem less severe for me.

I also did a test run with the idea i mentioned in the issue:
```
          const double error_now = std::fabs(actual_period.seconds() * write_rate - 1.0);
          const double error_skip = std::fabs(
            (actual_period.seconds() + 1.0 / resource_storage_->cm_update_rate_) * write_rate -
            1.0);
          if (error_now <= error_skip)
          {
            ret_val = component.write(current_time, actual_period);
          }
```
(and the same for read as well). The idea here is to figure out how much the target rate is violated if we do the read/write now vs skip it and do it in the next cycle. This seems to completely solve the problem for me:
```
  - key: kuka_kr50.read_cycle.execution_time
    value: 'Avg: 2.94 [0.36 - 44.31] us, StdDev: 2.95'
  - key: kuka_kr50.read_cycle.periodicity
    value: 'Avg: 250.04 [232.20 - 276.51] Hz, StdDev: 3.08 -> Desired : 250 Hz'
  - key: kuka_kr50.write_cycle.execution_time
    value: 'Avg: 3.91 [0.70 - 219.17] us, StdDev: 4.67'
  - key: kuka_kr50.write_cycle.periodicity
    value: 'Avg: 250.10 [217.62 - 284.97] Hz, StdDev: 5.02 -> Desired : 250 Hz'

Read skips: 3/58098 => 0.01%
Write skips: 4/58098 => 0.01%
Duration: 232.39s
```

[image]

Do you see a problem with this approach? Otherwise i'll try creating a full PR for this.

PR #2091 (2025-03-05): [RM] Fix skipped cycles by adjusting `rw_rate` handling
This PR fixes #2089 by changing the way `ResourceManager` decides if a hardware components `read()`/`write()` should be called in a give cycle. Previously, it only ensured that these don't get called more often than defined by `rw_rate`, leading to occasional skipped cycles in the presence of any jitter. This PR instead considers the "timing error" of either calling the components function in the cycle vs skipping it. I added a test that verifies the correct behavior for slightly shorter and slightly longer update rates.

I did some manua [... cut]