ros-controls/ros2_control #2104: Add Diagnostics to ros2_control

zkeyg, 2025-03-14:
**Is your feature request related to a problem? Please describe.**
Hi I am trying to set up a system with ros2_control. I met the issue https://github.com/ros-controls/ros2_control/issues/1185. I noticed that @bmagyar  mentioned that ros2_control has its own diagnotisc published to /diagnostics. However I went back to my setup and realized that I was using Ros2-humble. Diagnostics has been reverted (https://github.com/ros-controls/ros2_control/pull/1015) due to https://github.com/ros2/rclcpp/pull/2183. 

**Describe the solution you'd like**
Since https://github.com/ros2/rclcpp/pull/2183 has been merged, is it able to re-enable diagnsotics in ros2_control?

**Describe alternatives you've considered**

**Additional context**

bmagyar, 2025-03-14:
Oh, thanks for following up on this!
It is ABI breaking but I am sort of open to this... What do you gents think? @destogl @christophfroehlich @saikishor

christophfroehlich, 2025-03-14:
[gazebo_ros2_control](https://build.ros2.org/job/Hbin_uJ64__gazebo_ros2_control__ubuntu_jammy_amd64__binary/) still is built for humble? Otherwise it will break.

Raivias, 2025-07-01:
Humble doesn't EOL until 2027, shouldn't we still be supporting the gazebo for it?