ros-controls/ros2_control #2113: Logging to /rosout from hardware_interface.cpp

zacharyyamaoka, 2025-03-18:
Hi ROS2 Control team,

I was wondering if there is a way to log to /rosout from the hardware interface.cpp? For example:

```
hardware_interface::CallbackReturn MoteusHardwareInterface::on_init(const hardware_interface::HardwareInfo & info) {

  if (hardware_interface::SystemInterface::on_init(info) != hardware_interface::CallbackReturn::SUCCESS) {
    return hardware_interface::CallbackReturn::ERROR;
  }

    RCLCPP_INFO(rclcpp::get_logger("hardware_interface"), "Creating Member Functions");
```

This logs to terminal, but it doesn't show up in /rousout, so it doesn't show up in /rosout reading tools like Foxglove:

The controller manager is able to log to /rosout
```
[ INFO][4:07:38.982 PM PDT][bam_BAMGPU.controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ WARN][4:07:38.987 PM PDT][bam_BAMGPU.controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ INFO][4:07:39.777 PM PDT][bam_BAMGPU.controller_manager]: Loading controller 'joint_state_broadcaster'
[ INFO][4:07:39.801 PM PDT][bam_BAMGPU.controller_manager]: Configuring controller 'joint_state_broadcaster'
```

I belive this is beacuse it [uses a logger](https://github.com/ros-controls/ros2_control/blob/08f666cc272313c9c00c6102a59ff9b766ea4156/controller_manager/src/ros2_control_node.cpp#L103
) associated with its node:
```
  RCLCPP_INFO(cm->get_logger(), "update rate is %d Hz", cm->get_update_rate());
  const int thread_priority = cm->get_parameter_or<int>("thread_priority", kSchedPriority);
  RCLCPP_INFO(
    cm->get_logger(), "Spawning %s RT thread with scheduler priority: %d", cm->get_name(),
    thread_priority);
```

I have looked into it a bit and found:
Related issue: https://github.com/ros2/rclcpp/issues/2221
Related solution: https://github.com/ros2/rclcpp/pull/2233

```
  RCLCPP_INFO(node_logging_interface_->get_logger(), "Creating Moteus Driver");
```

Using that proposed solution doesn't seem to work though, 

Is there a standard reccomended way I could log to rosout?

thank you!

Zach

christophfroehlich, 2025-03-19:
Which distro are you using? From Jazzy on, the hardware components have a logger as child loggers from the CM
https://github.com/ros-controls/ros2_control_demos/blob/5bef8b48d66b5e673bf8ef63affa94f1049963ee/example_1/hardware/rrbot.cpp#L45
Can you try that?

zacharyyamaoka, 2025-03-29:
I was using Iron. I upgraded to Jazzy today, and replaced 
```
  RCLCPP_INFO(rclcpp::get_logger("hardware_interface"), "Loading Joint Params");
```
with
```
  RCLCPP_INFO(get_logger(), "Loading Joint Params");
```
It works now amazingly and shows up in foxglove! Thanks :)

[image]

closed 2025-03-29 by zacharyyamaoka