ros-controls/ros2_control #2116: Memory leaks detected by address sanitizer

fmauch, 2025-03-19:
**Describe the bug**
Running the tests using address sanitizer yields a couple of memory issues. I don't post the output as it is very long and can be easily reproduced using the steps below.

```
The following tests FAILED:
          1 - test_controller_manager (Failed)
          2 - test_controller_manager_with_namespace (Failed)
          3 - test_controller_manager_hardware_error_handling (Failed)
          4 - test_load_controller (Failed)
          5 - test_controllers_chaining_with_controller_manager (Failed)
          6 - test_controller_manager_srvs (Failed)
          7 - test_controller_manager_urdf_passing (Failed)
          8 - test_release_interfaces (Failed)
          9 - test_spawner_unspawner (Failed)
         10 - test_hardware_spawner (Timeout)
         11 - test_hardware_management_srvs (Failed)
         12 - test_ros2_control_node (Failed)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build ros2_control with asan enabled (e.g. by `--mixin asan-gcc`)
2. Disable new-delete-type-mismatch checks: `export ASAN_OPTIONS=new_delete_type_mismatch=0` (they are thrown from rclcpp and make using asan quite useless otherwise)
3. Run the tests using `colcon test --executor sequential --packages-select controller_interface controller_manager hardware_interface hardware_interface_testing joint_limits transmission_interface`
4. See error

**Expected behavior**
No issues raised by ASAN :-)

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version Jazzy / 08f666cc

**Additional context**
This may be potentially related to #1089

Mehul0x, 2025-04-01:
Hi! 
Is it possible for me to work on this too?

christophfroehlich, 2025-04-05:
Sure. First try to reproduce the errors, then let's discuss possible fixes if it is not obvious what to do..