ros-controls/ros2_control #2118: gazebo_ros2_control/GazeboSystem is missing

wahaha-1, 2025-03-20:
I plan to control a Mecanum wheel car (with imu, depth camera, lidar), but this error has not been resolved:
`[INFO] [launch]: All log files can be found below /home/wahaha/.ros/log/2025-03-20-14-34-12-393597-wahaha-Lenovo-Legion-Y7000P2021H-11534
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [11537]
[INFO] [gzclient-2]: process started with pid [11539]
[INFO] [robot_state_publisher-3]: process started with pid [11541]
[INFO] [spawn_entity.py-4]: process started with pid [11543]
[INFO] [ros2_control_node-5]: process started with pid [11545]
[INFO] [spawner-6]: process started with pid [11547]
[ros2_control_node-5] [WARN] [1742452452.999483447] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-5] [INFO] [1742452452.999994528] [resource_manager]: Loading hardware 'depth_camera_control'
[ros2_control_node-5] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[ros2_control_node-5]   what():  According to the loaded plugin descriptions the class gazebo_ros2_control/GazeboSystem with base class type hardware_interface::SystemInterface does not exist. Declared types are  fake_components/GenericSystem mock_components/GenericSystem test_hardware_components/TestSystemCommandModes test_hardware_components/TestTwoJointSystem
[ros2_control_node-5] Stack trace (most recent call last):
[ros2_control_node-5] #16   Object "", at 0xffffffffffffffff, in
[ros2_control_node-5] #15   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x5fd7fd897ee4, in
[robot_state_publisher-3] [INFO] [1742452453.017222679] [robot_state_publisher]: got segment back_shell_link
[robot_state_publisher-3] [INFO] [1742452453.017336199] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1742452453.017342737] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1742452453.017346803] [robot_state_publisher]: got segment depth_cam_frame
[robot_state_publisher-3] [INFO] [1742452453.017350670] [robot_state_publisher]: got segment depth_cam_link
[robot_state_publisher-3] [INFO] [1742452453.017354344] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1742452453.017357942] [robot_state_publisher]: got segment lidar_frame
[robot_state_publisher-3] [INFO] [1742452453.017362000] [robot_state_publisher]: got segment lidar_link
[robot_state_publisher-3] [INFO] [1742452453.017365777] [robot_state_publisher]: got segment lidar_sim_frame
[robot_state_publisher-3] [INFO] [1742452453.017369412] [robot_state_publisher]: got segment mic_link
[robot_state_publisher-3] [INFO] [1742452453.017372838] [robot_state_publisher]: got segment screen_camera_link
[robot_state_publisher-3] [INFO] [1742452453.017376566] [robot_state_publisher]: got segment screen_link
[robot_state_publisher-3] [INFO] [1742452453.017380066] [robot_state_publisher]: got segment speaker_link
[robot_state_publisher-3] [INFO] [1742452453.017383666] [robot_state_publisher]: got segment wheel_left_back_link
[robot_state_publisher-3] [INFO] [1742452453.017387574] [robot_state_publisher]: got segment wheel_left_front_link
[robot_state_publisher-3] [INFO] [1742452453.017391267] [robot_state_publisher]: got segment wheel_right_back_link
[robot_state_publisher-3] [INFO] [1742452453.017394865] [robot_state_publisher]: got segment wheel_right_front_link
[ros2_control_node-5] #14   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7b1200829e3f]
[ros2_control_node-5] #13   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7b1200829d8f]
[ros2_control_node-5] #12   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x5fd7fd8973e3, in
[ros2_control_node-5] #11   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7b12012ea5da, in controller_manager::ControllerManager::ControllerManager(std::shared_ptr<rclcpp::Executor>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&)
[ros2_control_node-5] #10   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7b12012e55ff, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ros2_control_node-5] #9    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7b1200ee3115, in hardware_interface::ResourceManager::load_urdf(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool)
[ros2_control_node-5] #8    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7b1200ee152e, in
[ros2_control_node-5] #7    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7b1200ec23fa, in
[ros2_control_node-5] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7b1200cae4d7, in __cxa_throw
[ros2_control_node-5] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7b1200cae276, in std::terminate()
[ros2_control_node-5] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7b1200cae20b, in
[ros2_control_node-5] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7b1200ca2b9d, in
[ros2_control_node-5] #2    Source "./stdlib/abort.c", line 79, in abort [0x7b12008287f2]
[ros2_control_node-5] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7b1200842475]
[ros2_control_node-5] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-5]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-5]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7b12008969fc]
[ros2_control_node-5] Aborted (Signal sent by tkill() 11545 1000)
[ERROR] [ros2_control_node-5]: process has died [pid 11545, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_667e9j2e --params-file /home/wahaha/Desktop/replay/install/gazebo_simulation/share/gazebo_simulation/config/jetauto_controllers.yaml --params-file /home/wahaha/Desktop/replay/install/gazebo_simulation/share/gazebo_simulation/config/depth_camera_controllers.yaml --params-file /home/wahaha/Desktop/replay/install/gazebo_simulation/share/gazebo_simulation/config/imu_controllers.yaml --params-file /home/wahaha/Desktop/replay/install/gazebo_simulation/share/gazebo_simulation/config/lidar_controllers.yaml --params-file /home/wahaha/Desktop/replay/install/gazebo_simulation/share/gazebo_simulation/config/screen_camera_controllers.yaml --params-file /tmp/launch_params_cioti5z3'].`

Here are the relevant files:

<?xml version="1.0" encoding="utf-8"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <ros2_control name="jetauto_base_control" type="system">
    <hardware>
      <plugin>gazebo_ros2_control/GazeboSystem</plugin>
    </hardware>
    
    <joint name="wheel_left_front_joint">
      <command_interface name="velocity">
        <param name="min">-10</param>
        <param name="max">10</param>
      </command_interface>
      <state_interface name="position"/>
      <state_interface name="velocity"/>
      <state_interface name="effort"/>
[... 283 lines cut ...]
                    executable='spawner',
                    arguments=['screen_camera_publisher', '--controller-manager', '/controller_manager'],
                    output='screen',
                )
            ]
        )
    )

    LaunchDescription
    return LaunchDescription([
        declare_use_sim_time,
        declare_world_file,
        declare_rviz,
        gazebo,
        robot_state_publisher,
        spawn_entity,
        control_node,
        joint_state_broadcaster_spawner,
        event_joint_state_broadcaster,
        imu_broadcaster_spawner,
        lidar_broadcaster_spawner,
        depth_camera_spawner,
        screen_camera_broadcaster_spawner,
        rviz
    ])

christophfroehlich, 2025-03-20:
gazebo classic reached end-of-life and is not maintained any more. It seems that you haven't installed the plugin correctly, can you load the gazebo_ros2_control demos?

closed 2025-03-20 by christophfroehlich

christophfroehlich, 2025-03-20:
The problem is
```
[ros2_control_node-5]   what():  According to the loaded plugin descriptions the class gazebo_ros2_control/GazeboSystem with base class type hardware_interface::SystemInterface does not exist. Declared types are  fake_components/GenericSystem mock_components/GenericSystem test_hardware_components/TestSystemCommandModes test_hardware_components/TestTwoJointSystem
```

it does not find your gazebo_ros2_control plugin. Have you installed and sourced it?

wahaha-1, 2025-03-20:
Yes, I confirmed that I installed：
```
wahaha@wahaha-Lenovo-Legion-Y7000P2021H:~$ dpkg -l | grep     ros-humble-ros2-control
ii  ros-humble-ros2-control                            2.48.0-1jammy.20250207.202533           amd64        Metapackage for ROS2 control related packages
ii  ros-humble-ros2-control-test-assets                2.48.0-1jammy.20250207.195608           amd64        The package provides shared test resources for ros2_control stack
ii  ros-humble-ros2-controllers                        2.41.0-1jammy.20250207.204752           amd64        Metapackage for ros2_controllers related packages
```
Do I need to install from source?
(Sorry for the late reply)

wahaha-1, 2025-03-20:
Here are the files that might be involved:
[control_depth_camera.txt](https://github.com/user-attachments/files/19364453/control_depth_camera.txt)
[control_jetauto.txt](https://github.com/user-attachments/files/19364456/control_jetauto.txt)
[gazebo_depth_camera.txt](https://github.com/user-attachments/files/19364454/gazebo_depth_camera.txt)
[jetauto_car.urdf.txt](https://github.com/user-attachments/files/19364455/jetauto_car.urdf.txt)
[jetauto_gazebo.launch.txt](https://github.com/user-attachments/files/19364457/jetauto_gazebo.launch.txt)

wahaha-1, 2025-03-20:
Yes, there are
`ros-humble-gazebo-ros2-control: /opt/ros/humble/lib/libgazebo_ros2_control.so
`
Here is the full information:
`ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/environment
ros-humble-gazebo-ros2-control: /opt/ros/humble/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/package.dsv
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/cmake/ament_cmake_export_include_directories-extras.cmake
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/local_setup.sh
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/cmake
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/cmake/gazebo_ros2_controlConfig-version.cmake
ros-humble-gazebo-ros2-control: /opt/ros/humble/lib/libgazebo_ros2_control.so
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/gazebo_hardware_plugins.xml
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/environment/library_path.sh
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/cmake/gazebo_ros2_controlConfig.cmake
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/ament_index/resource_index/package_run_dependencies/gazebo_ros2_control
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/local_setup.zsh
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/local_setup.bash
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/cmake/ament_cmake_export_dependencies-extras.cmake
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/environment/path.sh
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/package.xml
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/ament_index/resource_index/gazebo_ros2_control__pluginlib__plugin
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/ament_index/resource_index/gazebo_ros2_control__pluginlib__plugin/gazebo_ros2_control
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/ament_index/resource_index/parent_prefix_path/gazebo_ros2_control
ros-humble-gazebo-ros2-control: /opt/ros/humble/include/gazebo_ros2_control/gazebo_system_interface.hpp
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/local_setup.dsv
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/environment/path.dsv
ros-humble-gazebo-ros2-control: /opt/ros/humble/include/gazebo_ros2_control
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/environment/ament_prefix_path.dsv
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/environment/ament_prefix_path.sh
ros-humble-gazebo-ros2-control: /opt/ros/humble/include/gazebo_ros2_control/gazebo_system.hpp
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/environment/library_path.dsv
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/gazebo_ros2_control/cmake/ament_cmake_export_libraries-extras.cmake
ros-humble-gazebo-ros2-control: /opt/ros/humble/share/ament_index/resource_index/packages/gazebo_ros2_control
`

wahaha-1, 2025-03-20:
我刚刚检查了一下：在这个路径下有如下文件和定义：
`/opt/ros/humble/share/gazebo_ros2_control/gazebo_hardware_plugins.xml`

gazebo_hardware_plugins.xml

```
<library path="gazebo_hardware_plugins">
  <class
    name="gazebo_ros2_control/GazeboSystem"
    type="gazebo_ros2_control::GazeboSystem"
    base_class_type="gazebo_ros2_control::GazeboSystemInterface">
    <description>
      GazeboPositionJoint
    </description>
  </class>
</library>
```
But when I run `ros2 component types | grep -i gazebo_ros2_control`, there is no output
并且运行`ros2 component types `

[types.txt](https://github.com/user-attachments/files/19367361/types.txt)