ros-controls/ros2_control #2119: [ros2_control_node-1] [ERROR] [1742473810.665745003] [hand_controller]: Joint name cannot be empty

mynamelzy, 2025-03-20:
my  yaml is 
`controller_manager:
  ros__parameters:
    update_rate: 250  # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    forward_command_controller_position:
      type: forward_command_controller/ForwardCommandController

    hand_controller:
      type: position_controllers/GripperActionController

hand_controller:
  ros__parameters:
    type: position_controllers/GripperActionController
    joints:
      - right_joint
    command_interfaces:
      - position
    state_interfaces:
      - position

joint_trajectory_controller:
  ros__parameters:
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6
    command_interfaces:
      - position
      - velocity
    state_interfaces:
      - position
      - velocity
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.0
      goal_time: 0.0

forward_command_controller_position:
  ros__parameters:
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6
    interface_name: position`

and  my hand.ros2_control.xacro
`<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:macro name="hand_ros2_control" params="name prefix">
    <ros2_control name="${name}" type="system">
      <hardware>
          <plugin>topic_based_ros2_control/TopicBasedSystem</plugin>
          <param name="joint_commands_topic">/hand_joint_commands</param>
          <param name="joint_states_topic">/hand_joint_states</param>
      </hardware>
      <joint name="right_joint">
        <command_interface name="position"/>
        <state_interface name="position">
        <param name="initial_value">0.0</param>
        </state_interface>
      </joint>
    </ros2_control>
  </xacro:macro>

</robot>`

my yaml has right_joint,but it shows me "Joint name cannot be empty",i dont know how to solove this question

christophfroehlich, 2025-03-20:
Can you post the full log please?

mynamelzy, 2025-03-21:
> Can you post the full log please?

OK, this is the log 
`[INFO] [launch]: All log files can be found below /home/liu/.ros/log/2025-03-20-20-30-10-046331-liu-107794
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [107797]
[INFO] [robot_state_publisher-2]: process started with pid [107799]
[INFO] [spawner-3]: process started with pid [107801]
[robot_state_publisher-2] [INFO] [1742473810.349891716] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1742473810.350052979] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1742473810.350077423] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1742473810.350094953] [robot_state_publisher]: got segment left_Link
[robot_state_publisher-2] [INFO] [1742473810.350114998] [robot_state_publisher]: got segment link_1
[robot_state_publisher-2] [INFO] [1742473810.350128896] [robot_state_publisher]: got segment link_2
[robot_state_publisher-2] [INFO] [1742473810.350144121] [robot_state_publisher]: got segment link_3
[robot_state_publisher-2] [INFO] [1742473810.350158439] [robot_state_publisher]: got segment link_4
[robot_state_publisher-2] [INFO] [1742473810.350173664] [robot_state_publisher]: got segment link_5
[robot_state_publisher-2] [INFO] [1742473810.350187911] [robot_state_publisher]: got segment link_6
[robot_state_publisher-2] [INFO] [1742473810.350203276] [robot_state_publisher]: got segment right_Link
[robot_state_publisher-2] [INFO] [1742473810.350217524] [robot_state_publisher]: got segment shelf_link
[robot_state_publisher-2] [INFO] [1742473810.350232889] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1742473810.350247555] [robot_state_publisher]: got segment tool0_link
[robot_state_publisher-2] [INFO] [1742473810.350262432] [robot_state_publisher]: got segment world
[ros2_control_node-1] [WARN] [1742473810.381953967] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1742473810.382328454] [resource_manager]: Loading hardware 'ABBMultiInterfaceHardware' 
[ros2_control_node-1] [INFO] [1742473810.383228842] [resource_manager]: Initialize hardware 'ABBMultiInterfaceHardware' 
[ros2_control_node-1] [WARN] [1742473810.383261667] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-1] [INFO] [1742473810.383300708] [resource_manager]: Successful initialization of hardware 'ABBMultiInterfaceHardware'
[ros2_control_node-1] [INFO] [1742473810.383325851] [resource_manager]: Loading hardware 'hand_controller' 
[ros2_control_node-1] [INFO] [1742473810.384670007] [resource_manager]: Initialize hardware 'hand_controller' 
[ros2_control_node-1] [INFO] [1742473810.393462829] [resource_manager]: Successful initialization of hardware 'hand_controller'
[ros2_control_node-1] [INFO] [1742473810.393544682] [resource_manager]: 'configure' hardware 'hand_controller' 
[ros2_control_node-1] [INFO] [1742473810.393555368] [resource_manager]: Successful 'configure' of hardware 'hand_controller'
[ros2_control_node-1] [INFO] [1742473810.393564726] [resource_manager]: 'activate' hardware 'hand_controller' 
[ros2_control_node-1] [INFO] [1742473810.393574714] [resource_manager]: Successful 'activate' of hardware 'hand_controller'
[ros2_control_node-1] [INFO] [1742473810.393581977] [resource_manager]: 'configure' hardware 'ABBMultiInterfaceHardware' 
[ros2_control_node-1] [INFO] [1742473810.393605653] [resource_manager]: Successful 'configure' of hardware 'ABBMultiInterfaceHardware'
[ros2_control_node-1] [INFO] [1742473810.393617596] [resource_manager]: 'activate' hardware 'ABBMultiInterfaceHardware' 
[ros2_control_node-1] [INFO] [1742473810.393626745] [resource_manager]: Successful 'activate' of hardware 'ABBMultiInterfaceHardware'
[ros2_control_node-1] [INFO] [1742473810.400072705] [controller_manager]: update rate is 250 Hz
[ros2_control_node-1] [INFO] [1742473810.400122851] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1742473810.400383706] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-1] [INFO] [1742473810.629873913] [controller_manager]: Loading controller 'hand_controller'
[spawner-3] [INFO] [1742473810.664211648] [spawner_hand_controller]: Loaded hand_controller
[ros2_control_node-1] [INFO] [1742473810.665446224] [controller_manager]: Configuring controller 'hand_controller'
[ros2_control_node-1] [INFO] [1742473810.665685359] [hand_controller]: Action status changes will be monitored at 20Hz.
[ros2_control_node-1] [ERROR] [1742473810.665745003] [hand_controller]: Joint name cannot be empty
[ros2_control_node-1] [WARN] [1742473810.665795498] [hand_controller]: Error occurred while doing error handling.
[ros2_control_node-1] [ERROR] [1742473810.665855631] [controller_manager]: After configuring, controller 'hand_controller' is in state 'unconfigured' , expected inactive.
[spawner-3] [ERROR] [1742473810.666683105] [spawner_hand_controller]: Failed to configure controller
[ERROR] [spawner-3]: process has died [pid 107801, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner hand_controller -c /controller_manager --ros-args'].

`

mynamelzy, 2025-03-21:
I see position_controllers/GripperActionController use the joint ,not the joints

closed 2025-03-21 by mynamelzy