ros-controls/ros2_control #2121: controller_interface::ChainableControllerInterface exported_state_interfaces_[name]->set_value() throws exception

ul-l-i, 2025-03-20:
**Describe the bug**
Using exported_state_interfaces_[name]->set_value() always results in an error:

[ros2_control_node-2] Stack trace (most recent call last) in thread 242414:
[ros2_control_node-2] #13   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-2] #12   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 78, in clone3 [0x75ae97b29c3b]
[ros2_control_node-2] #11   Source "./nptl/pthread_create.c", line 447, in start_thread [0x75ae97a9caa3]
[ros2_control_node-2] #10   Source "../../../../../src/libstdc++-v3/src/c++11/thread.cc", line 104, in execute_native_thread_routine [0x75ae97eecdb3]
[INFO] [spawner-16]: process has finished cleanly [pid 242606]
[ros2_control_node-2] #9    Object "/opt/ros/jazzy/lib/controller_manager/ros2_control_node", at 0x5b2a4451f43e, in 
[ros2_control_node-2] #8    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x75ae98664bb7, in controller_manager::ControllerManager::update(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-2] #7    Object "/opt/ros/jazzy/lib/libcontroller_interface.so", at 0x75ae981f405e, in controller_interface::ControllerInterfaceBase::trigger_update(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-2] #6    Object "/home/leipscheru/ros2_ws/install/drive_controller/lib/libodometry_provider.so", at 0x75ae534143c2, in four_wheel_steering::OdometryProvider::update_and_write_commands(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-2] #5    Object "/home/leipscheru/ros2_ws/install/drive_controller/lib/libodometry_provider.so", at 0x75ae534328be, in bool hardware_interface::Handle::set_value<double>(double const&)
[ros2_control_node-2] #4    Object "/home/leipscheru/ros2_ws/install/drive_controller/lib/libodometry_provider.so", at 0x75ae53439d6f, in std::unique_lock<std::shared_mutex>::unique_lock(std::shared_mutex&, std::try_to_lock_t)
[ros2_control_node-2] #3    Object "/home/leipscheru/ros2_ws/install/drive_controller/lib/libodometry_provider.so", at 0x75ae53417ff3, in std::shared_mutex::try_lock()
[ros2_control_node-2] #2    Object "/home/leipscheru/ros2_ws/install/drive_controller/lib/libodometry_provider.so", at 0x75ae53417e77, in std::__shared_mutex_pthread::try_lock()
[ros2_control_node-2] #1    Object "/home/leipscheru/ros2_ws/install/drive_controller/lib/libodometry_provider.so", at 0x75ae53417dd3, in std::__glibcxx_rwlock_trywrlock(pthread_rwlock_t*)
[ros2_control_node-2] #0    Source "./nptl/pthread_rwlock_trywrlock.c", line 37, in ___pthread_rwlock_trywrlock [0x75ae97aa2fe4]
[ros2_control_node-2] Segmentation fault (Address not mapped to object [0x80])
[ERROR] [ros2_control_node-2]: process has died [pid 242339, exit code -11, cmd '/opt/ros/jazzy/lib/controller_manager/ros2_control_node --ros-args --log-level info --ros-args --params-file /tmp/launch_params_hc57lyvr'].

**Expected behavior**
A working function call without exception.

**Screenshots**

[image]

**Environment (please complete the following information):**
 - OS: Ubuntu24.04
 - Version : Jazzy

ul-l-i, 2025-03-21:
I did and the outcome is the same. Just to be sure, I want to share my "on_export_state_interfaces" method implementation:

`std::vector<hardware_interface::StateInterface> OdometryProvider::on_export_state_interfaces()
{
  state_interfaces_values_.clear();
  state_interfaces_values_.resize(6, 0.0);
  std::vector<hardware_interface::StateInterface> state_interfaces;
  state_interfaces.push_back(hardware_interface::StateInterface(
      get_node()->get_name(), interface_name0, &state_interfaces_values_[0]));
  state_interfaces.push_back(hardware_interface::StateInterface(
      get_node()->get_name(), interface_name1, &state_interfaces_values_[1]));
  state_interfaces.push_back(hardware_interface::StateInterface(
      get_node()->get_name(), interface_name2, &state_interfaces_values_[2]));
  state_interfaces.push_back(hardware_interface::StateInterface(
      get_node()->get_name(), interface_name3, &state_interfaces_values_[3]));
  state_interfaces.push_back(hardware_interface::StateInterface(
      get_node()->get_name(), interface_name4, &state_interfaces_values_[4]));
  state_interfaces.push_back(hardware_interface::StateInterface(
      get_node()->get_name(), interface_name5, &state_interfaces_values_[5]));

  return state_interfaces;
}`

Two more points:

1.  `exported_state_interfaces_[name]` exists, but neither `->get_value()` nor ` ->get_optional()` works
 
2. `state_interfaces_values_[0] = value;` works without problems

saikishor, 2025-03-23:
I tried with example_12 from ros2_control_demos with reference interface and it would get it working

```
  for (size_t i = 0; i < command_interfaces_.size(); ++i)
  {
    const std::string ref_name = std::string(get_node()->get_name()) + "/" + reference_interface_names_[i];
    const double val = exported_reference_interfaces_[ref_name]->get_optional().value();
    if (!std::isnan(val))
    {
      RCLCPP_WARN(
        get_node()->get_logger(), "Setting command %s to %f", command_interface_names_[i].c_str(),
        val);
      command_interfaces_[i].set_value(val);
    }
  }
```

Can you get a minimal example to reproduce the issue?

ul-l-i, 2025-03-24:
Hey, never mind. 
The issue is solved, it was the interface_name being incomplete. I did not add the node name to it.
Though before the last update it worked without the node name perfectly fine. Are such changes communicated somewhere? That would be great.

Thanks for looking into it.

saikishor, 2025-03-24:
> Hey, never mind. 
> The issue is solved, it was the interface_name being incomplete. I did not add the node name to it.
> Though before the last update it worked without the node name perfectly fine. Are such changes communicated somewhere? That would be great.
> 
> Thanks for looking into it.

Hello!

You are right. This was recently introduced in https://github.com/ros-controls/ros2_control/commit/465e5d0264dcaca71858022ec907a85cba45f811. Unfortunately, this is needed as if you have 2 different joints with the same interfaces, it won't work at all. This is the reason, this change was introduced. I hope this answers your question. I'll go ahead and close the issue.

saikishor, 2025-03-24:
> @saikishor can we add this to the migration notes? We merged this kind of breaking change in Jazzy.

Yess!

closed 2025-03-24 by christophfroehlich