ros-controls/ros2_control #2127: ROS Topic when the controller is starting -  on_init or on_activate

brennand, 2025-03-23:
System: ROS 2 Jazzy

Hi,
I'm writing a custom hardware interface for an **EtherCAT** system. It works well, but it takes around 30–60 seconds for the system to fully initialize.

I'm wondering — is there a way to use **/dynamic_joint_states** (or another topic) to publish status updates during `on_init()` or `on_activate()`?

It would be really helpful if **/dynamic_joint_states** could start publishing at that stage.

Reason:
When I launch the system from the command line, I can see what's happening through the console output. But when running on the real robot, I use a GUI that monitors the internal system state using custom **/dynamic_joint_states** messages — and during startup, it’s silent.

I’m wondering if there’s a better or more standard way to expose what’s happening during initialization?

Bren

christophfroehlich, 2025-03-23:
During on_init() this is not possible. But in inactive state all the exported state and command interfaces are available, we recently updated [the docs](https://control.ros.org/jazzy/doc/ros2_control/hardware_interface/doc/hardware_components_userdoc.html) to make that clearer.
This means that any ros2_controllers can publish the information even if the hardware component is not in active state.

bmagyar, 2025-03-23:
`dynamic_joint_states` are published by the broadcaster, not the hardware component. 

Just to clarify: your goal is to monitor what's happening during startup, not to use that information elsewhere, right? To that end we've recently added some new support to [`pal_statistics`](https://github.com/ros-controls/ros2_control/pull/1918) and also scoped up a GSoC project to generally improve logging and introspection there: https://github.com/osrf/osrf_wiki/wiki/GSoC25#hardware-diagnostics-support-for-ros2_control

brennand, 2025-03-24:
Thank you for your fast feedback.  I will have a look today.