ros-controls/ros2_control #2130: Robustify the XML parser and don't crash on missing plugin

christophfroehlich, 2025-03-24:
## Background

As reported [here](https://github.com/ros-controls/ros2_control/issues/2128), the system crashes if there is no plugin in the ros2_control section of the URDF.

Add a try-catch block here or check if `type_it` is valid, and give a proper error message otherwise

https://github.com/ros-controls/ros2_control/blob/def08f0b01b328dc8d8b3312f8094e1d64986b57/hardware_interface/src/component_parser.cpp#L662-L664

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

Mehul0x, 2025-03-25:
Hi! I am a first-time contributor; can I be allotted this?

saikishor, 2025-03-25:
@Mehul0x This issue is assigned to you now :)

Mehul0x, 2025-03-26:
I have made the changes as required to both instances where type_it is being used, how do I check if my change works and that it doesn't break anything?

christophfroehlich, 2025-03-26:
From https://github.com/ros-controls/ros2_control/blob/master/CONTRIBUTING.md: 

> Ensure local tests pass. (colcon test and pre-commit run (requires you to install pre-commit by pip3 install pre-commit)

Mehul0x, 2025-03-26:
where do I run the colcon test?

in ~/ros2_ws/ ? or inside the ros2_ws/src/ros-controls/ros2_control ?

christophfroehlich, 2025-03-26:
From your workspace, see https://docs.ros.org/en/rolling/Tutorials/Intermediate/Testing/CLI.html and other documentation of `colcon test`

Mehul0x, 2025-03-26:
I got this as an output. Are test case failing in unrelated packages a problem? the package, the change was made to passed the test cases, so I guess its ok? 

```
aerosheesh-24@aerosheesh-24-ASUS-TUF-Gaming-A15-FA506QM-FA506QM:~/ros2_ws$ colcon test --ctest-args tests
Starting >>> control_msgs
Starting >>> realtime_tools
Starting >>> ros2_control_test_assets
Starting >>> controller_manager_msgs
Starting >>> ros2_control_demo_description
Starting >>> ros2_controllers_test_nodes
Starting >>> kinematics_interface
Finished <<< ros2_control_test_assets [0.21s]                                                                                                          
Finished <<< ros2_control_demo_description [0.20s]
Finished <<< kinematics_interface [0.20s]
Starting >>> kinematics_interface_kdl
Finished <<< kinematics_interface_kdl [0.31s]                                                                                                                                           
Finished <<< ros2_controllers_test_nodes [13.2s]                                                                                                           
Finished <<< realtime_tools [15.3s]                                                                                  
Starting >>> joint_limits
Finished <<< joint_limits [1.43s]                                                                                 
Finished <<< controller_manager_msgs [26.0s]                                                
Finished <<< control_msgs [53.2s]             
Starting >>> hardware_interface
Starting >>> control_toolbox
Starting >>> rqt_joint_trajectory_controller
Finished <<< rqt_joint_trajectory_controller [0.51s]                                                                             
Finished <<< control_toolbox [2.19s]                                                     
Finished <<< hardware_interface [2.56s]             
Starting >>> controller_interface
Starting >>> hardware_interface_testing               
Starting >>> transmission_interface
Finished <<< transmission_interface [1.17s]                                                                                           
Finished <<< controller_interface [1.61s]                                                             
Finished <<< hardware_interface_testing [6.87s]                 
Starting >>> controller_manager
[Processing: controller_manager]                          
[Processing: controller_manager]                             
[Processing: controller_manager]                               
[Processing: controller_manager]                               
[Processing: controller_manager]                               
[Processing: controller_manager]                               
[Processing: controller_manager]                               
[Processing: controller_manager]                               
[Processing: controller_manager]                               
[Processing: controller_manager]                               
Finished <<< controller_manager [5min 11s]                     
Starting >>> joint_state_broadcaster
Starting >>> ros2controlcli
Starting >>> forward_command_controller                                                           
Starting >>> joint_trajectory_controller
Starting >>> steering_controllers_library
Starting >>> diff_drive_controller
Starting >>> force_torque_sensor_broadcaster
Starting >>> gz_ros2_control
Starting >>> gpio_controllers
Starting >>> imu_sensor_broadcaster
Starting >>> mecanum_drive_controller
Starting >>> tricycle_controller
Starting >>> pid_controller
Starting >>> admittance_controller
Starting >>> gripper_controllers
Starting >>> parallel_gripper_controller
Finished <<< steering_controllers_library [1.11s]                                                                                                                               
Starting >>> ackermann_steering_controller
Finished <<< ros2controlcli [1.42s]                                                                                                                                             
Starting >>> bicycle_steering_controller
Finished <<< imu_sensor_broadcaster [1.67s]                                                                                                                                            
Starting >>> pose_broadcaster
Finished <<< gz_ros2_control [2.46s]                                                                                                                                                   
Starting >>> range_sensor_broadcaster
[... 142 lines cut ...]
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
---
Finished <<< ros2_control_demo_example_1 [36.6s]	[ with test failures ]
Starting >>> ros2_control_demo_example_15
--- stderr: gz_ros2_control_tests                                                                                                                       
Errors while running CTest
Output from these tests are in: /home/aerosheesh-24/ros2_ws/build/gz_ros2_control_tests/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
---
Finished <<< gz_ros2_control_tests [45.0s]	[ with test failures ]
--- stderr: ros2_control_demo_example_12                                                                                
Errors while running CTest
[... cut ...] le_9
  17 packages had test failures: diff_drive_controller gpio_controllers gz_ros2_control_tests ros2_control_demo_example_1 ros2_control_demo_example_10 ros2_control_demo_example_11 ros2_control_demo_example_12 ros2_control_demo_example_14 ros2_control_demo_example_16 ros2_control_demo_example_2 ros2_control_demo_example_3 ros2_control_demo_example_4 ros2_control_demo_example_5 ros2_control_demo_example_6 ros2_control_demo_example_7 ros2_control_demo_example_8 ros2_control_demo_example_9

```

christophfroehlich, 2025-03-26:
use `colcon test --packages-select hardware_interface` to only test the relevant package. but it is strange that so many are failing. nevermind, open a PR and we'll see.

Mehul0x, 2025-03-26:
That has passed, I have opened a PR. 
I hope I fixed the issue :)

closed 2025-03-28 by christophfroehlich