ros-controls/ros2_control #2140: Project Idea: Add support for hardware semantic components

christophfroehlich, 2025-03-26:
The ros2_control framework relies on simple command and state interfaces in the form of double values to exchange data between hardware components and controllers. It is desired however to provide good C++ data structures both on the hardware component and the controller side which improves code readability and maintainability.

Earlier in the project, the concept of semantic components were introduced which essentially provides a grouping for these values and a semantic-specific API to use them. For instance, an IMU sensor will typically report 9 values, 3 values for each axis of the accelerometer, gyroscope and compass sensor parts respectively. Such values can be grouped and served through an API as a ROS IMU message or as a C++ struct for both input and output of these values while the ros2_control framework maintains it's low-profile communication interfaces internally.

The goal of this project is to add semantic components that are relevant for hardware components.
Additionally, this project includes extending the existing simulation tools with a set of common semantic components to support different sensors and actuators.

Related design document and code implementations:

* [Few idea about Semantic Components for hardware](https://github.com/ros-controls/roadmap/pull/45)
* [Semantic Components for Controllers](https://github.com/ros-controls/ros2_control/tree/master/controller_interface/include/semantic_components)

----

| Skills required/preferred:

* Good C++ skills
* Basic understanding of ROS and/or ROS 2
* Basic understanding of the Gazebo simulator

| Possible mentors: Bence Magyar, Denis Štogl
| Expected size of project: 175 hours
| Difficulty: medium

christophfroehlich, 2025-03-28:
First we should start the discussion in https://github.com/ros-controls/roadmap/pull/45 and make a design about how that should look like. Drop your comments there, and we can discuss it in the next PMC meeting.

imsudiproy, 2025-03-28:
Hi @christophfroehlich ,
Just to let you know, I don't have any hands-on experience with ROS or the Gazebo simulator yet. I'm taking this as a learning opportunity. So it would be better to assign it to someone who has more experience with it, while I'll continue working on it locally. I'd appreciate it if I get the exact requirements, so that make progress on it from my side locally.