ros-controls/ros2_control #2141: Project Idea: Hardware Diagnostics Support for ros2_control

christophfroehlich, 2025-03-26:
- Prerequisites: git and Github
-    Necessary programming skills: strong C++, ROS 2
-    Difficulty level: Medium/Hard
-    Potential mentors: [Bence Magyar](https://github.com/bmagyar), [Sai Kishor](https://github.com/saikishor/)
-    Expected size: 175 hours to 350 hours
-    Expected outcome: Reporting capabilities are available for hardware components with minimal code changes

## Detailed description
The ros2_control framework uses a plugin-based system to support multiple hardware drivers at the same time. It takes care of resource constraints between different controllers and hardware components, asynchronous operation of controllers, etc. As of now, controller_manager publishes the diagnostics of itself, and for controllers and hardware components about their lifecycle state and operational statistics such as their periodicity and their execution time. However, A current limitation of the framework is that hardware components, e.g. the ros2_control hardware driver plugin for a given robot, doesn't have a good API for reporting the diagnostics of its own hardware, for instance, the state of the CAN Bus, some internal motor control board stats like temperature, error codes etc. This integration would make the ros2_control, provide complete diagnostics within its ecosystem. This project focuses on setting up an API that hardware manufacturers & driver maintainers can use to report things with minimal code changes to their existing setups.

## Goals

-  Modify the Hardware Interfaces to fetch the hardware diagnostics.
-  Possible unification of three hardware interfaces into a single one.
-  Add tests to test the integration.
-  Add Documentation.

Nandostream11, 2025-05-26:
Hi @christophfroehlich ,

I had drafted a GSoC proposal for this exact project earlier, and although I couldn’t submit it within the final deadline, I remain deeply interested in contributing to this effort.

In my proposal, I outlined a plan to extend ros2_control’s diagnostics to provide robust hardware-level observability. The goal is to design a minimally intrusive, extensible API enabling hardware interface plugins to report diagnostics such as CAN bus status, IMU data, motor driver error codes, temperature, voltage levels, etc., all using the existing diagnostic_updater stack. 
A concise plan based on my earlier GSoC proposal:
1. Define a lightweight diagnostics API for hardware interfaces using diagnostic_updater.
2. Add diagnostics hooks in Actuator, Sensor, and System interfaces (possibly would try unifying them).
4. Implementation of example usage in a demo hardware plugin.
5. Perform integration tests for diagnostics reporting.
6. Adding documentation and tutorials.

I’ve worked on relevant projects like quadruped robot sensor integration, swarm navigation systems, and a [PR](https://github.com/process1183/ros2_bno055/pull/1) for the BNO055 IMU plugin, and have hands-on experience with ROS 2. I’d be grateful for any guidance from the mentors & maintainers, especially on aligning my implementation plan with the upstream expectations. If this issue is still open for contribution, I would love to begin working on it and submit a formal PR.

saikishor, 2025-05-26:
> Hi @Nandostream11! Afaik we have a GSoC candidate working on this, right @saikishor @bmagyar?

Yes, we have a GSoC candidate @soham2560 working on this