ros-controls/ros2_control #2144: Remap hardware interface topics

Tacha-S, 2025-03-28:
Lately, I’ve been seeing logs suggesting the use of `--controller-ros-args` for remapping, so I’d like to update my setup accordingly.

However, my hardware interface publishes some topics that need to be remapped.  
I’m currently passing remap rules when launching `ros2_control_node`,  
but how exactly does `--controller-ros-args` handle remapping for topics published by the hardware interface?

christophfroehlich, 2025-03-28:
>  but how exactly does `--controller-ros-args` handle remapping for topics published by the hardware interface?

not at all. You manually create the node in your hardware component, right?
There exists a hardware_spawner, but it does not know your node so we can't add something similar to  `--controller-ros-args` there without giving it manually also the node name, am I right @saikishor?

saikishor, 2025-03-29:
> not at all. You manually create the node in your hardware component, right?
> There exists a hardware_spawner, but it does not know your node so we can't add something similar to [...]

@christophfroehlich You are right. Right now, there is no way to remap them apart through the CM. Morever, we don't recommend this type of integration suffers with same node name issue when the ros2_control_node is named (https://github.com/ros-controls/ros2_control/issues/1684)

christophfroehlich, 2025-03-29:
As adding a node into the hardware_component is a recurring pattern: Should we add this per default so that users don't struggle with breaking any RT loops? A little bit related to #2141

Tacha-S, 2025-03-31:
> You manually create the node in your hardware component, right?

Yes.

Currently, the remapping is working as expected.  
Therefore, as long as the warning about `--controller-ros-args`—that only `--controller-ros-args` will be used and the usual node options will be ignored when starting the controller—does not affect the current remap functionality of topics within the hardware components, there is no particular issue.

> A little bit related to https://github.com/ros-controls/ros2_control/issues/2141

I also think being able to report diagnostics is a good.

saikishor, 2025-03-31:
> As adding a node into the hardware_component is a recurring pattern: Should we add this per default so that users don't struggle with breaking any RT loops? A little bit related to #2141 

Maybe yes. We can discuss in the PMC meeting

MCFurry, 2025-07-21:
Ahh I'm bumping into the same issue here while trying to remove all warnings from an ongoing project.
We also have created a node from within our hardware_interface and are not able to remap topics except from remapping them on the controller manager. Which in turn throws us this warning of course:
`The use of remapping arguments to the controller_manager node is deprecated. Please use the '--controller-ros-args' argument of the spawner to pass remapping arguments to the controller node.`

If I know my hardware_interface node's name, can I then remap from the spawner? I guess not since the spawner has nothing to do with the SystemInterface plugin?

Can I safely ignore the remapping warning for now? Or is this likely to break in a future release?

christophfroehlich, 2025-07-24:
@soham2560 is working on that topic, we have now a default node for the hardware plugins. How can we handle (and document) the remapping?

soham2560, 2025-07-24:
Yes, @MCFurry you can refer [here](https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/writing_new_hardware_component.html#writing-a-hardware-component) for the documentation on how to use the default node capability.

For handling the default hardware node remapping I think 
- we should accept a remap arg through the <ros2_control> URDF tag, basically get it through `HardwareInfo` 
or
- create something along the lines of `--harware-ros-args`, that takes the remapping through `ResourceManagerParams` down to `init`?

@saikishor opinions?

closed 2025-10-23 by github-actions[bot]

christophfroehlich, 2025-10-24:
@soham2560 was there a decision on your questions, is this solved already?

VivekVRobo, 2026-09-24:
Hi, I’d like to work on a focused part of this issue.

I reviewed the current hardware-node initialization path before proposing a change.

`ControllerManager::init_resource_manager()` passes the robot description, executor, namespace, clock, and other configuration through `ResourceManagerParams`, but there is currently no ROS/node-arguments field propagated through `ResourceManagerParams` → `HardwareComponentParams`.

I also noticed that `HardwareComponentInterface::init()` currently does:

```cpp
auto options = define_custom_node_options();
options.arguments({"--ros-args", "-r", "__node:=" + node_name});
```

so the hardware component node is ultimately created with the generated node-name remap, without an explicit path for user-supplied hardware-node ROS arguments.

Rather than trying to address every remapping/deprecation concern in one large PR, I’d like to start with the hardware-node argument path.

My proposed scope is:

1. Add an explicit hardware-node ROS/node-options argument path through the ResourceManager/HardwareComponent parameter structs.
2. Propagate the relevant arguments when ControllerManager initializes hardware components.
3. Merge those arguments with the generated hardware node name rather than accidentally discarding existing custom `NodeOptions` arguments.
4. Add focused tests showing that a remap reaches the hardware component’s default ROS node.
5. Document the supported behavior.

I would leave the `--hardware-ros-args` hardware-spawner UX and removal of legacy controller-manager remapping/deprecation behavior for a follow-up unless you would prefer those changes together.

Does this split match the direction you had in mind?