ros-controls/ros2_control #2147: Add tests for launch_utils of controller manager

saikishor, 2025-03-28:
## Background

We have nice utilities for spawner launching of controllers, however, there are no tests or no usage of them in ros2_control_demos to have them tested via functional tests that we already have. 

It would be nice to have tests in place for these launch_utils to avoid situations like in PR : #2146.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

Ayushman-Saha, 2025-03-29:
Hi,
I'm a 2nd year student with a keen interest in robotics and I'm looking to get started with open source contributions. I saw this "good-first-issue" and it seems like a great opportunity. I'd like to claim it and work on it.

Regards
Ayushman Saha

Mehul0x, 2025-03-30:
Hi, 
Can I help with this too? 

I suppose [ROS2-Testing](https://docs.ros.org/en/humble/Tutorials/Intermediate/Testing/Testing-Main.html) has enough information to do this?

saikishor, 2025-03-30:
> Hi, 
> Can I help with this too? 
> 
> I suppose [ROS2-Testing](https://docs.ros.org/en/humble/Tutorials/Intermediate/Testing/Testing-Main.html) has enough information to do this?

@Mehul0x Unfortunately, we can only assign it to one person. As @Ayushman-Saha approached this first, he gets to work on it and probably he is already on it

Ayushman-Saha, 2025-03-31:
@saikishor Sorry, the setting up of local workspace took longer than I expected. Should I start including my tests in the existing test_utils.py file or should I create a separate file for the tests?

christophfroehlich, 2025-11-01:
But you haven't updated your PR?