ros-controls/ros2_control #2151: Add consistent transitions for controllers in `rqt_controller_manager`

soham2560, 2025-03-29:
## Brief
This issue is in context to a converstion at https://github.com/ros-controls/ros2_control/pull/2143, where @christophfroehlich pointed out that the transitions for the controllers are not consistent (refer https://github.com/ros-controls/ros2_control/pull/2143#pullrequestreview-2726474276)

Currently in `rqt_controller_manager` you can alter the state of the controller from one to another, but some transitions are missing in some states, notably
- unload transition from unconfigured
- deactivate->unload transition from active?
and more

A potential solution is to do as suggested in https://github.com/ros-controls/ros2_control/pull/2143#issuecomment-2763312004
> .......basically every controller should be in one of 4 states (atleast in context of `rqt_controller_manager`)
    - active
    - inactive
    - unconfigured
    - unloaded
    and you should have 3 options at each state to go to the other ones
    Each transition consists of a seqential combination of elements  of either
      `Load` -> `Configure`-> `Activate`
        or
      `Deactivate`-> `Cleanup`-> `Unload`

This is up for discussion however, since too many options can also be confusing, and you can always switch one by one between the states

soham2560, 2025-03-29:
@christophebedard though about what you said in https://github.com/ros-controls/ros2_control/pull/2143#issuecomment-2763367187, I think the most minimal transition setup you can have will be to have the following options in each state
- `active` - [`deactivate`]
- `inactive` - [`activate,cleanup`]
- `unconfigured` - [`configure,unload`]
- `unloaded` - [`load`]

I am leaning more towards the minimal method, because the all transitions method does seem like it will create more confusion

christophebedard, 2025-03-29:
I assume you meant to tag @christophfroehlich

soham2560, 2025-03-29:
Yes! apologies for the incorrect tag, have updated, thanks for letting me know

christophfroehlich, 2025-03-29:
I agree with the minimal solution, but IMHO we should stick to the "official" transition wording from https://design.ros2.org/articles/node_lifecycle.html

- We don't have the `finalized` state, and hence no `create`/`destroy` transition. Instead, we have the` on_init()` callback for creation, but none for destruction. I'm unsure if we should stay with `load` name, or use `load/create` to have both worlds?
- Let's add `unload/shutdown` transition from every state, calling the unload_controller service (which itself calls the shutdown transition of the node from the CM).
- To make that foolproof for users not familiar with the lifecycle transitions (like me, I always get confused with cleanup and shutdown), we could add the target state in parentheses?

@saikishor what do you think?

saikishor, 2025-03-29:
> -  We don't have the finalized state, and hence no create/destroy transition. Instead, we have the on_init() callback for creation, but none for destruction. I'm unsure if we should stay with load name, or use load/create to have both worlds?

I think it is better to stick with the controller terminology of `load` alone

> -  Let's add unload/shutdown transition from every state, calling the unload_controller service (which itself calls the shutdown transition of the node from the CM).

@christophfroehlich Do you also mean with `active` state?

> -  To make that foolproof for users not familiar with the lifecycle transitions (like me, I always get confused with cleanup and shutdown), we could add the target state in parentheses?

I don't know if it better to include lifecycle transitions here. For me, it is clear from the controller states. Moreover, controllers use the lifecycle node, but they are not completely lifecycle node, so for this reason I'm not sure

christophfroehlich, 2025-03-30:
> I think it is better to stick with the controller terminology of `load` alone

ok

> > * Let's add unload/shutdown transition from every state, calling the unload_controller service (which itself calls the shutdown transition of the node from the CM).
> 
> [@christophfroehlich](https://github.com/christophfroehlich) Do you also mean with `active` state?

I see that we don't allow this in the CM, so we could call inactive -> unload in this case.

> I don't know if it better to include lifecycle transitions here. For me, it is clear from the controller states. Moreover, controllers use the lifecycle node, but they are not completely lifecycle [...]

We list the _tranistions_ as text, and the color code gives the target _state_. What I meant, let's also write the target _state_ there, like `🟢  activate (active)`, or `🟡 cleanup (unconfigured)`

saikishor, 2025-03-30:
> We list the _tranistions_ as text, and the color code gives the target _state_. What I meant, let's also write the target _state_ there, like `🟢 activate (active)`, or `🟡 cleanup (unconfigured)`

Sure. 👍🏾

soham2560, 2025-03-31:
Sorry for the late reply

> To make that foolproof for users not familiar with the lifecycle transitions (like me, I always get confused with cleanup and shutdown), we could add the target state in parentheses?

this makes sense to me as well, since new users would also prefer to use the GUI and this would make it easier for them

> We list the transitions as text, and the color code gives the target state. What I meant, let's also write the target state there, like 🟢 activate (active), or 🟡 cleanup (unconfigured)

Sounds good! I do feel the wording can be better, for me it does not give the clear impression of "_this transition would lead to this state_" upon first read, maybe because the words for transition and state are sometimes similar, sometimes not, but I think we can atleast set it up, the words can be changed later

Now we will have following transitions at each stage
- `active` 🟢
    - 🔵 deactivate (inactive)
    - ⚫ deactivate and unload (unloaded)
- `inactive` 🔵
    - 🟢 activate (active)
    - 🟡 cleanup (unconfigured)
    - ⚫ unload (unloaded)
- `unconfigured` 🟡
    - 🔵 configure (inactive)
    - ⚫ unload (unloaded)
- `unloaded` ⚫
    - 🟡 load (unconfigured)

Why am I getting the feeling this also looks a bit inconsistent? Let me know what your opinion is

saikishor, 2025-03-31:
It probably should be something like the following:

- `active` 🟢
    - 🔵 deactivate (inactive)
    - ⚫ deactivate and unload (unloaded)
- `inactive` 🔵
    - 🟢 activate (active)
    - ⚫ unload (unloaded)
- `unconfigured` 🟡
    - 🟢 configure & activate (active)
    - 🔵 configure (inactive)
    - ⚫ unload (unloaded)
- `unloaded` ⚫
    - 🟡 load (configured)

I'm still not sure, if we should support the `unloaded` option, in future it shouldn't work this way. Right now, it is working as some cleanup is missing inside the CM.

soham2560, 2025-03-31:
- `activate` transition from unconfigured makes sense
- Why remove the `cleanup` transition from inactive? wouldn't this mean that to go to an unconfigured state the user would have to manually unload and load again? Any reason we can't do it internally?

soham2560, 2025-04-01:
@saikishor @christophfroehlich any opinions on the above?

soham2560, 2025-04-03:
@saikishor @christophfroehlich I've added a draft PR at #2163 so that we can atleast get down to a visual representation, do check it out
The logic in the PR is
- `active` 🟢
    - 🔵 deactivate (inactive)
    - ⚫ deactivate and unload (unloaded)
- `inactive` 🔵
    - 🟢 activate (active)
    - 🟡 cleanup (unconfigured)
    - ⚫ unload (unloaded)
- `unconfigured` 🟡
    - 🟢 configure and activate (active)
    - 🔵 configure (inactive)
    - ⚫ unload (unloaded)
- `unloaded` ⚫
    - 🟡 load (unconfigured)

It is basically what @saikishor suggested in https://github.com/ros-controls/ros2_control/issues/2151#issuecomment-2765263895 but with `cleanup` transition in `inactive` state, as I think that should also be there

but do give me your opinion!

Also, to be technically right we can rename the `cleanup` transition to `unload and load`

soham2560, 2025-04-05:
@christophfroehlich @saikishor let me know how it looks

saikishor, 2025-04-05:
For me, it looks good. I'm just not sure about unload and load operation. I don't have a strong opinion here, if everyone is ok with it. Let's do it

PR #2163 (2025-04-17): Add consistent transitions for controllers in rqt_controller_manager
## Brief
In this PR
- Adding consistent transitions to `rqt_controller_manager` (refer 546c9c0b4594880eb6048b01e40ecc9472231246)
   - Fixes #2151 (do refer the conversation at the issue for a more detailed discussion) 

### How was this tested?
- Launch ros2_control_demo_example_14 with
    ```bash
  ros2 launch ros2_control_demo_example_14 rrbot_modular_actuators_without_feedback_sensors_for_position_feedback.launch.py
  ```
- Launch rqt_controller_manager with
   ```bash
  ros2 run rqt_controller_manager rqt_controller_mana [... cut]
closed 2025-04-17 by christophfroehlich