ros-controls/ros2_control #2158: Missing ros2_control_cmake

aalmrad, 2025-04-01:
**Describe the bug**
When building `ros2_control` from source (latest `master` branch) for ROS 2 Jazzy, `rosdep` fails with:
Cannot locate rosdep definition for [ros2_control_cmake]

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the master branch of the ros2_control repository to the /src directory in your ros2 workspace
2. Use the following command to fetch the needed dependencies: rosdep install --from-paths src --ignore-src -r -y

**Expected behavior**
rosdep should successfully install all the required rosdeps

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version: ROS2 Jazzy

**Additional context**
I have to manually clone the  ros2_control_cmake repository in my /src directory. Is this expected?

saikishor, 2025-04-01:
If I'm not wrong, rosdep installs the synced packages. As ros2_control_cmake is something new and doesn't exist in Jazzy sync. It is compiling. Better clone it for now and compile. This should be fixed post Jazzy sync

christophfroehlich, 2025-04-03:
Please follow the [installation guideline](https://control.ros.org/jazzy/doc/getting_started/getting_started.html#building-from-source) and use the correct repos file to download the dependencies.

closed 2025-04-03 by christophfroehlich