ros-controls/ros2_control #2161: Source build trys to install rosdeps

christophfroehlich, 2025-04-03:
**Describe the bug**

```

  ERROR: the following rosdeps failed to install
    apt: command [apt-get install -y ros-rolling-diagnostic-updater] failed
    apt: Failed to detect successful installation of [ros-rolling-diagnostic-updater]

```
**To Reproduce**
https://github.com/ros-controls/ros2_control/actions/runs/14236642887/job/39897093251

**Expected behavior**
Build the package from source, not installing it ;) 

I think we have to update the packages in a repos file which are not part of [ros2.repos](https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos)

fmauch, 2025-04-03:
Isn't that simply failing because the package is currently [failing in the buildfarm ](https://github.com/ros/diagnostics/issues/457)and thus doesn't end up in the testing repo?

christophfroehlich, 2025-04-03:
yes that's why the job is failing now, but isn't the purpose of the source builds tu build _everyhing_ from source? ;)

closed 2025-04-16 by christophfroehlich