ros-controls/ros2_control #2171: micro-ros support

bmagyar, 2025-04-09:
**Is your feature request related to a problem? Please describe.**
Better support for boards such as

**Describe the solution you'd like**
Investigate potentially supporting micro-ros-based hardware OR provide documentation for going about this

**Describe alternatives you've considered**
https://github.com/PickNikRobotics/topic_based_ros2_control

yassine-cherni, 2025-04-09:
Hi @bmagyar  thanks for opening this issue, it’s spot-on for our work at [OORB-Open-Organic-Robotics](https://github.com/OORB-Open-Organic-Robotics) our diff-drive robot uses an RPi5 for high-level tasks (sensor fusion, navigation) and an STM32F4 Discovery for low-level control (encoders, IMU, motor PWM via PID), linked via a new SPI hardware_interface with micro-ROS.

Why It Matters
Micro-ROS support in ros2_control would simplify integrating microcontrollers like the STM32F4 Discovery ( or ESP32 ) into ROS 2, keeping real-time performance intact unlike topic_based_ros2_control, which drops that edge.

Proposed Solution
I’d suggest:

- A ros2_control template for micro-ROS hardware interfaces (e.g., SPI setup for STM32).

- Docs on building real-time micro-ROS plugins and optimizing latency aiming for <2 ms end-to-end latency would be ideal for embedded setups.

gavanderhoorn, 2025-04-09:
> Docs on building real-time micro-ROS plugins and optimizing latency aiming for <2 ms end-to-end latency would be ideal for embedded setups.

this would be great, but I don't expect this to be possible. At least not with standard TCP or even UDP based transports.

And with the Agent between your micro-ROS client and your ROS 2 / DDS application.

Especially for reliable topics, but even for best-effort topics (see eProsima/Micro-XRCE-DDS-Client#394) the overhead of having the additional hop through the Agent is quite significant.

tonybaltovski, 2025-04-10:
We use topics for the [Jackal's diff drive interface](https://github.com/clearpathrobotics/clearpath_robot/blob/jazzy/clearpath_hardware_interfaces/src/diff_drive/hardware_interface.cpp).  It has a micro-ROS based MCU currently.

bmagyar, 2025-04-10:
@yassine-cherni I know I asked already but are you using a specific board that has the stm32 integrated with the raspi or they are separate? Could you please share the exact link for it? (I know you did already at the meeting yesterday but I cannot recover it :-/)

bmagyar, 2025-04-10:
Thanks for the feedback @gavanderhoorn , insightful as always! Anything else you have on this front would be very valuable

yassine-cherni, 2025-04-10:
Hi @bmagyar , The STM32F4 Discovery and Raspberry Pi 5 are separate boards in our setup. We connect them via SPI, [Raspberry pi 5](https://www.raspberrypi.com/products/raspberry-pi-5/) , [Stm32f4 discovery](https://www.st.com/en/evaluation-tools/stm32f4discovery.html)

christophfroehlich, 2025-04-11:
I answered similar questions on stack exchange from time to time. Which means, there is a community interest (or confusion) about it. 

My question is, why do you need the extra layer of micro-ros? I can imagine that it provides some out-of-the box experience with supported uCs, e.g. without the hassle of writing SPI drivers in C++. But as already mentioned, I'm not sure if the desired realtime requirements can be met. But compared to topic_based_ros2_control there could still be room for improvement.