ros-controls/ros2_control #2172: ONNX in a controller

bmagyar, 2025-04-09:
**Is your feature request related to a problem? Please describe.**
Support the ONNX runtime in a controller

**Describe the solution you'd like**
Questionable, if this is a requirement for the framework or more for a demo.

bmagyar, 2025-04-09:
FYI @olivier-stasse if you have any more details to add

qiayuanl, 2025-04-12:
ONNX runtime can be easily added as a custom dependency in some learning-based controllers. What is the relationship between it and the ros-control framework?

Juliaj, 2025-06-26:
@qiayuanl , IMO, ONNX runtime is largely transparent to the ROS2 Control Framework which defines standardized interfaces that controllers must implement, regardless of their internal implementation details. 

That said, there are a few practical considerations which developers may need to consider for learning based controller, for example: the performance overhead (including ONNX Runtime overhead) and how it fits within control loop timing constraints, error handling and real-time compliance etc.

closed 2026-11-04 by christophfroehlich