ros-controls/ros2_control #2185: Passing Parameter Values to Controllers

jcarpinelli-bdai, 2025-04-16:
The `robot_description` parameter is used in controllers, but users have no method of providing this value except using deprecated behavior. **How should users provide the `robot_description` parameter to controllers in newer ROS distros?**

Previously, the `robot_description` parameter was passed directly to the `controller_manager`, and the `controller_manager` set this parameter for all loaded controllers. This is how controllers which require robot kinematics, e.g. Cartesian controllers, retrieved the robot description. 

Providing the `robot_description` via ROS parameters is now deprecated in the `controller_manager`. A warning states that users should instead let the `controller_manager` read from the `/robot_description` topic. If users provide the robot description through the description topic, how should they pass this value to controllers? Is the new expected behavior for users to subscribe to the robot description topic within controllers?

saikishor, 2025-04-16:
Hello @jcarpinelli-bdai!

You can now access the robot description directly from within the controllers using the `get_robot_description` method. The controllers get the robot_description from the controller manager.

https://github.com/ros-controls/ros2_control/blob/65340a435a9f2e59a077e168c3daaf1db1dc0581/controller_interface/src/controller_interface_base.cpp#L280

I hope this answers your question.

jcarpinelli-bdai, 2025-04-16:
Thank you @saikishor! That does answer my question, but now I am curious about other parameter values. For example, a `semantic_robot_description`. It feels like users may benefit from having a way to set parameters outside of a parameter file. Could the parameters of the spawner node be forwarded to each controller?

saikishor, 2025-04-16:
@jcarpinelli-bdai this can be achieved using the `--controller-ros-args` option of the spawner

jcarpinelli-bdai, 2025-04-16:
That's great, thank you! Apologies for the null issue then :)

saikishor, 2025-04-23:
> It appears this is not supported on Humble. What is the supported method for passing parameter values to spawned controllers in Humble?
> 
> @saikishor @christophfroehlich 

That's right. Humble doesn't support this feature.