ros-controls/ros2_control #2189: docs: What are preceding and following controllers?

peci1, 2025-04-16:
There is this paragraph in the chaining docs:

https://github.com/ros-controls/ros2_control/blob/de00c17e643b19e7980e522965f23cf22c0a9c21/controller_manager/doc/controller_chaining.rst?plain=1#L95-L100

It says that all following controllers need to be activated before their preceding controllers.

I don't see a good definition of preceding, so I'll use my intuition here. If I look at https://github.com/ros-controls/roadmap/blob/master/design_drafts/controller_chaining.md#example-2 , `pid left wheel` is preceding `diff_drive_controller` and `diff_drive_controller` follows `pid left wheel`. This is how I understand it.

If my understanding is correct, then the docs are wrong (at least on Jazzy). I wrote an example using a very similar chain to the above-mentioned example: https://github.com/ros-controls/gz_ros2_control/issues/306#issuecomment-2810895120 .

If I start the `diff_drive_controller` first, it fails to activate. If I start the wheel PIDs first, then everything works fine.

christophfroehlich, 2025-04-17:
> I don't see a good definition of preceding, so I'll use my intuition here. If I look at https://github.com/ros-controls/roadmap/blob/master/design_drafts/controller_chaining.md#example-2 , `pid left [...]

It is the other way round, "following" is the later one in the forward-chain. In the linked example, preceding controller is the diff_drive_controller, the PIDs are bot following controllers. Do you mind adding a PR in clarifying things in the docs?

peci1, 2025-04-17:
Sure, I'll send a PR. So the ordering is from the outermost controller that receives commands from subscribers down to the hardware interfaces?

christophfroehlich, 2025-04-17:
no wait, the activation order has to be from the innermost (following) controller, because it first has to export the reference (command) interfaces, so that the preceding controller can claim them.

PR #2192 (2025-04-17): doc: Added explanation of preceding/following controllers
Resolves #2189 .

This PR adds explanation of what preceding/following controller means.

I also broke the rule and attached an irrelevant fix of grammar in the paragraph directly *preceding* the paragraph I added. I hope it's not a big no no :)
closed 2025-04-17 by christophfroehlich