ros-controls/ros2_control #2226: SwitchController ordering

Petrigh, 2025-05-14:
Hi, I am trying to switch a predecessor controller that should chain or unchain with [diff_drive_controller](https://control.ros.org/rolling/doc/ros2_controllers/diff_drive_controller/doc/userdoc.html#references) on command, as the documentation says:

> The chained controllers can also be activated when parsed as in a single list through the fields activate_controllers or deactivate_controllers in the switch_controllers service provided by the controller_manager.

Now I am setting a switch_controllers request with activate_controllers and deactivate_controllers set as I need, but the ordering will try to activate a controller (that is already active, I just want to maintain it once the predecessor is deactivated) and the try to deactivate the predecessor, making my diff_drive_controller get some .NAN inputs.

```
auto request = std::make_shared<SwitchCtrl::Request>();
request->deactivate_controllers = {controller_name};
request->activate_controllers   = controllers_to_unchain_;
request->strictness = SwitchCtrl::Request::BEST_EFFORT;
request->activate_asap = true;
request->timeout.sec = 5;

// Wait for service to be available
while (!switch_client_->wait_for_service(std::chrono::seconds(1))) {
  if (!rclcpp::ok()) {
    RCLCPP_ERROR(get_logger(), "Interrupted while waiting for the switch controller service. Exiting.");
    return;
  }
  RCLCPP_INFO(get_logger(), "Waiting for switch controller service...");
}

switch_client_->async_send_request(
  request,
  [this, controller_name](rclcpp::Client<SwitchCtrl>::SharedFuture resp_fut) {
    if (resp_fut.get()->ok) {
      RCLCPP_INFO(get_logger(), "Stopped %s", controller_name.c_str());
    } else {
      RCLCPP_ERROR(get_logger(), "Failed to witch controller service to stop %s", controller_name.c_str());
    }
  });
```

Is there a way to order the switching?
Here is my log:

```
[ros2_control_node-1] [INFO] [controller_manager]: Activating controllers: [ diff_drive_controller ]
[ros2_control_node-1] [INFO] [controller_manager]: Deactivating controllers: [ can_vel_controller ]
[ros2_control_node-1] [WARN] [controller_manager]: Controller with name 'diff_drive_controller' is already active.
[ros2_control_node-1] [WARN] [controller_manager]: Could not activate controller with name 'diff_drive_controller'. Check above warnings for more details.
[ros2_control_node-1] [INFO] [controller_manager]: Activated controllers: [ diff_drive_controller ]
[ros2_control_node-1] [INFO] [controller_manager]: Successfully switched controllers!
[ros2_control_node-1] [WARN] [diff_drive_controller]: Command message contains NaNs. Not updating reference interfaces.
[teleop_node-6] [INFO][controller_toggle_node]: Stopped can_vel_controller
```

Moreover, once the switching is done, my diff drive controller is not subscribed to cmd_vel.

saikishor, 2025-05-14:
Hello @Petrigh 

> Now I am setting a switch_controllers request with activate_controllers and deactivate_controllers set as I need, but the ordering will try to activate a controller (that is already active, I just [...]

I'm not sure if I understand your scenario. Do you mind explaining with an example?. What do you mean by ordering the switching?

In the logs, you say that the `can_vel_controller` continues to be active?

Petrigh, 2025-05-14:
Hi @saikishor , thanks for the quick replay.
I started to get confused with myself, and so many changes made me think there was an issue with the switchcontroller, my diff_drive states a warning that says
`[ros2_control_node-1] [WARN] [1747221306.874553991] [omnitract_motor_driver]: Command message contains NaNs. Not updating reference interfaces.`

But for the moment is nothing that makes my project halt.
Thanks again for the attention and quick response.
The issue is solved, i just called the switch controller without any activate_controllers  when trying to turn off can_vel_controller

```
    auto request = std::make_shared<SwitchCtrl::Request>();
    request->deactivate_controllers = {controller_name};
    request->strictness = SwitchCtrl::Request::BEST_EFFORT;
    request->activate_asap = true;
    request->timeout.sec = 5;
    
    switch_client_->async_send_request()
...
```

saikishor, 2025-05-14:
Thank you. Good to know.

closed 2025-05-14 by saikishor