ros-controls/ros2_control #2234: bool data_type of hardware interfaces specified in urdf file is ignored

alexanderkorepov, 2025-05-16:
I have many true/false command and state interfaces. When bool type was added as a possible value type I decided to use it instead of converting values to double. But I found I can't use them as my interfaces are generated based on urdf files and they all are generated as double interfaces.

This is the interface from urdf file:
`<state_interface name="enabled" data_type="bool"/>`

I set the value:
`command_interfaces_[0].set_value(false);`

And get the error:
`Invalid data type : 'bool' access for interface : hardware/enabled expected : 'double'`

I looked at `ros2_control` sources and found that if the type passed to `set_value` is not `double` then it's compared with the type of `value_` in `Handle` and if it's not the same it throws the exception I see.
```
    if constexpr (std::is_same_v<T, double>)
    {
      // If the template is of type double, check if the value_ptr_ is not nullptr
      THROW_ON_NULLPTR(value_ptr_);
      *value_ptr_ = value;
    }
    else
    {
      if (!std::holds_alternative<T>(value_))
      {
        throw std::runtime_error(
          "Invalid data type : '" + get_type_name<T>() + "' access for interface : " + get_name() +
          " expected : '" + data_type_.to_string() + "'");
      }
      value_ = value;
    }
```

So, the value type of `Handle` can be set only during the construction and cannot be changed later.

This is how the interface is constructed from urdf file:
```
    component.state_interfaces.push_back(parse_interfaces_from_xml(state_interfaces_it));
    component.state_interfaces.back().data_type = parse_data_type_attribute(state_interfaces_it);
```
`parse_interfaces_from_xml` creates an interface with `double` type, next line sets `data_type` to `bool` but it's too late, `Handle` is already holding `double` value.

saikishor, 2025-05-16:
Hello @alexanderkorepov!

Thank you for reporting the issue. We will take a look and try to fix it sooner.

Best Regards,
ros2_control Team

saikishor, 2025-05-16:
Hello!

Looking at the code, it seems to be only supported by the GPIOs, but not for sensors and joints. Can you confirm that your case is not GPIOs?
https://github.com/ros-controls/ros2_control/blob/ad2efc6fb8adb89c2b3af0d55280e1901ee46fa3/hardware_interface/src/component_parser.cpp#L693-L696

Thank you

saikishor, 2025-05-16:
@alexanderkorepov The PR: 2235 should fix it

alexanderkorepov, 2025-05-16:
> Looking at the code, it seems to be only supported by the GPIOs, but not for sensors and joints. Can you confirm that your case is not GPIOs?
I'm using GPIOs

closed 2025-05-21 by christophfroehlich