ros-controls/ros2_control #2236: Throttled overrun prints don't seem to be throttled properly

bmagyar, 2025-05-16:
**Describe the bug**
Throttling the warnings about overruns doesn't seem to work reliably.
We may need to explore using a different clock

**Expected behavior**
There should be at most 1 message per second.
The RCLCPP_WARN_THROTTLED print [here](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/ros2_control_node.cpp#L165) should throttle the messages to 1000ms = 1 second. 

**Screenshots**
[WARN] [1747395233.440807556] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 500 Hz. The loop took 2.895488 ms (missed cycles : 2).
[WARN] [1747395233.450173719] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 500 Hz. The loop took 2.263731 ms (missed cycles : 2).
[WARN] [1747395234.114839063] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 500 Hz. The loop took 2.927795 ms (missed cycles : 2).
[WARN] [1747395234.134072793] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 500 Hz. The loop took 2.161909 ms (missed cycles : 2).
[WARN] [1747395235.112376698] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 500 Hz. The loop took 2.466102 ms (missed cycles : 2).
[WARN] [1747395235.122050666] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 500 Hz. The loop took 2.139334 ms (missed cycles : 2).

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version Jazzy 4.29.0
- Docker container

**Additional context**
No custom changes

InigoMoreno, 2025-05-21:
Experiencing the same issue, it is almost impossible to read the logs.

saikishor, 2025-05-21:
> Experiencing the same issue, it is almost impossible to read the logs.

@InigoMoreno what distro are you using? And at what frequency are you running your controller_manager?

SteMuc, 2025-06-13:
Hi, I'm experiencing the same issue.
`[ros2_control_node-1] [WARN] [1749827164.547201688] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 50 Hz. The loop took 76.312262 ms (missed cycles : 4).`
Has the problem been solved? 

Thank you very much.

saikishor, 2025-06-13:
> Hi, I'm experiencing the same issue.
> `[ros2_control_node-1] [WARN] [1749827164.547201688] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 50 Hz. The loop took 76.312262 ms (missed cycles : 4).`
> Has the problem been solved? 
> 
> Thank you very much.

Hello @SteMuc !

You have the throttling issue? Or it's just printing it?
The logs you shared don't give much information about it.

luvwinnie, 2025-07-11:
how can we solve this problem?

dbarsoum, 2025-08-11:
Hi, I am also having the same issue as well (using ROS 2 Jazzy). I have tried lowering the frequency and this also did not help; I have tried 150, 100, 50, and 10 Hz. I still continue to get the error `Overrun detected! The controller manager missed its desired rate...`
The way I am lowering the the frequency is by running in the terminal `ros2 param set /controller_manager update_rate 10`
I am also looking at the system manager and don't think it's cpu related. I'm using linux and running ros through docker (Ubuntu 24.04). Any help with this would be great---thank you!

dbarsoum, 2025-08-12:
I tried one more thing and it seemed to fix it; @christophfroehlich was correct that a lower frequency needs to be set. I went into my robot's `controller_manager` package into the yaml file and lowered it from 150 (which was the issue) to 50. When I edited my post above, I only tried 100 and 10 Hz. This is the yaml file I changed: https://github.com/xArm-Developer/xarm_ros2/blob/humble/xarm_controller/config/xarm7_controllers.yaml under parameter `update_rate`. So far no issues. Hope this helps others! Thank you!