ros-controls/ros2_control #2243: `TestLoadController` spawner tests fail

christophfroehlich, 2025-05-17:
Latest [rolling source builds](https://github.com/ros-controls/ros2_control/actions/runs/15060150725) fail with 

```
2025-05-17T04:33:47.5386343Z     [  FAILED  ] TestLoadController.spawner_test_type_in_param
2025-05-17T04:33:47.5386520Z     [  FAILED  ] TestLoadController.multi_ctrls_test_type_in_param
2025-05-17T04:33:47.5386758Z     [  FAILED  ] TestLoadController.spawner_test_failed_activation_of_controllers
2025-05-17T04:33:47.5387064Z     [  FAILED  ] TestLoadControllerWithNamespacedCM.multi_ctrls_test_type_in_param
2025-05-17T04:33:47.5387132Z     
2025-05-17T04:33:47.5387209Z      4 FAILED TESTS
2025-05-17T04:33:47.5387317Z     -- run_test.py: return code 1
2025-05-17T04:33:47.5387951Z     -- run_test.py: inject classname prefix into gtest result file '/__w/ros2_control/ros2_control/ros_ws/build/controller_manager/test_results/controller_manager/test_spawner_unspawner.gtest.xml'
2025-05-17T04:33:47.5388479Z     -- run_test.py: verify result file '/__w/ros2_control/ros2_control/ros_ws/build/controller_manager/test_results/controller_manager/test_spawner_unspawner.gtest.xml'
2025-05-17T04:33:47.5388546Z   >>>
2025-05-17T04:33:47.5389070Z build/controller_manager/test_results/controller_manager/test_spawner_unspawner.gtest.xml: 26 tests, 0 errors, 4 failures, 0 skipped
2025-05-17T04:33:47.5389815Z - controller_manager.TestLoadController spawner_test_type_in_param (/__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:131)
2025-05-17T04:33:47.5389890Z   <<< failure message
2025-05-17T04:33:47.5390454Z     /__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:157
2025-05-17T04:33:47.5390573Z     Expected equality of these values:
2025-05-17T04:33:47.5390726Z       call_spawner("ctrl_1 -c test_controller_manager")
2025-05-17T04:33:47.5390799Z         Which is: 256
2025-05-17T04:33:47.5390863Z       0
2025-05-17T04:33:47.5391252Z     /__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:164
2025-05-17T04:33:47.5391362Z     Expected equality of these values:
2025-05-17T04:33:47.5391473Z       ctrl_1.c->get_lifecycle_state().id()
2025-05-17T04:33:47.5391554Z         Which is: '\x2' (2)
2025-05-17T04:33:47.5391725Z       lifecycle_msgs::msg::State::PRIMARY_STATE_ACTIVE
2025-05-17T04:33:47.5391807Z         Which is: '\x3' (3)
2025-05-17T04:33:47.5391869Z   >>>
2025-05-17T04:33:47.5392686Z - controller_manager.TestLoadController multi_ctrls_test_type_in_param (/__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:199)
2025-05-17T04:33:47.5392773Z   <<< failure message
2025-05-17T04:33:47.5393158Z     /__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:244
2025-05-17T04:33:47.5393269Z     Expected equality of these values:
2025-05-17T04:33:47.5393435Z       call_spawner("ctrl_1 ctrl_3 -c test_controller_manager")
2025-05-17T04:33:47.5393506Z         Which is: 256
2025-05-17T04:33:47.5393573Z       0
2025-05-17T04:33:47.5393946Z     /__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:246
2025-05-17T04:33:47.5394052Z     Expected equality of these values:
2025-05-17T04:33:47.5394168Z       cm_->get_loaded_controllers().size()
2025-05-17T04:33:47.5394237Z         Which is: 2
2025-05-17T04:33:47.5394308Z       3ul
2025-05-17T04:33:47.5394377Z         Which is: 3
2025-05-17T04:33:47.5394439Z   >>>
2025-05-17T04:33:47.5395261Z - controller_manager.TestLoadController spawner_test_failed_activation_of_controllers (/__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:467)
2025-05-17T04:33:47.5395397Z   <<< failure message
2025-05-17T04:33:47.5395776Z     /__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:556
2025-05-17T04:33:47.5395887Z     Expected equality of these values:
2025-05-17T04:33:47.5396370Z       call_spawner( "ctrl_with_joint1_and_joint2_command_interfaces -c test_controller_manager " "--controller-manager-timeout 1.0 " "-p " + test_file_path)
2025-05-17T04:33:47.5396443Z         Which is: 256
2025-05-17T04:33:47.5396507Z       0
2025-05-17T04:33:47.5396886Z     Should pass as the ctrl_with_joint1_command_interface and ctrl_with_joint2_command_interface are inactive
2025-05-17T04:33:47.5396960Z   >>>
2025-05-17T04:33:47.5397795Z - controller_manager.TestLoadControllerWithNamespacedCM multi_ctrls_test_type_in_param (/__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:973)
2025-05-17T04:33:47.5397875Z   <<< failure message
2025-05-17T04:33:47.5398261Z     /__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:1023
2025-05-17T04:33:47.5398367Z     Expected equality of these values:
2025-05-17T04:33:47.5398646Z       call_spawner("ctrl_1 ctrl_3 -c test_controller_manager --ros-args -r __ns:=/foo_namespace")
2025-05-17T04:33:47.5398715Z         Which is: 256
2025-05-17T04:33:47.5398777Z       0
2025-05-17T04:33:47.5399164Z     /__w/ros2_control/ros2_control/ros_ws/src/mj7g0exbxi/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:1026
2025-05-17T04:33:47.5399270Z     Expected equality of these values:
2025-05-17T04:33:47.5399391Z       cm_->get_loaded_controllers().size()
2025-05-17T04:33:47.5399461Z         Which is: 2
2025-05-17T04:33:47.5399600Z       3ul
2025-05-17T04:33:47.5399671Z         Which is: 3
2025-05-17T04:33:47.5399734Z   >>>
```

saikishor, 2025-05-17:
This is something from rclpy side, it is failing as the warn attribute is not found

```
2025-05-16T04:39:20.7946029Z 9: [INFO] [1747370360.793540942] [spawner_ctrl_1]: waiting for service /test_controller_manager/list_controllers to become available...
2025-05-16T04:39:20.8574301Z 9: Traceback (most recent call last):
2025-05-16T04:39:20.8578242Z 9:   File "/__w/ros2_control/ros2_control/ros_ws/install/controller_manager/lib/controller_manager/spawner", line 33, in <module>
2025-05-16T04:39:20.8579747Z 9:     sys.exit(load_entry_point('controller-manager==4.29.0', 'console_scripts', 'spawner')())
2025-05-16T04:39:20.8583441Z 9:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-05-16T04:39:20.8585322Z 9:   File "/__w/ros2_control/ros2_control/ros_ws/install/controller_manager/lib/python3.12/site-packages/controller_manager/spawner.py", line 239, in main
2025-05-16T04:39:20.8586573Z 9:     node.get_logger().warn(
2025-05-16T04:39:20.8586975Z 9:     ^^^^^^^^^^^^^^^^^^^^^^
2025-05-16T04:39:20.8587757Z 9: AttributeError: 'RcutilsLogger' object has no attribute 'warn'
```

saikishor, 2025-05-17:
Related to https://github.com/ros2/rclpy/pull/1456/files#diff-5f8e1845255f66261770ad7f33a557e848d3ebe7fc329fccf7c1269f93c81b5c

saikishor, 2025-05-17:
Will be fixed by #2244

closed 2025-05-17 by christophfroehlich