ros-controls/ros2_control #2250: Generic system Mock components does not work well with chained interfaces.

InigoMoreno, 2025-05-19:
I have tried to use the "Passthrough controller" of example 12 with a ur robot in simulation.  The passthrough controller correctly generates the interfaces:

```
        passthrough_controller/elbow_joint/position [available] [unclaimed]
        passthrough_controller/shoulder_lift_joint/position [available] [unclaimed]
        passthrough_controller/shoulder_pan_joint/position [available] [unclaimed]
        passthrough_controller/wrist_1_joint/position [available] [unclaimed]
        passthrough_controller/wrist_2_joint/position [available] [unclaimed]
        passthrough_controller/wrist_3_joint/position [available] [unclaimed]
```

When trying to connect a separate controller to these interfaces, however I get the following error:

```
[ros2_control_node-7] Joint 'shoulder_pan_joint' has to have 'position', 'velocity', or 'acceleration' interface!
[ros2_control_node-7] Joint 'shoulder_lift_joint' has to have 'position', 'velocity', or 'acceleration' interface!
[ros2_control_node-7] Joint 'elbow_joint' has to have 'position', 'velocity', or 'acceleration' interface!
[ros2_control_node-7] Joint 'wrist_1_joint' has to have 'position', 'velocity', or 'acceleration' interface!
[ros2_control_node-7] Joint 'wrist_2_joint' has to have 'position', 'velocity', or 'acceleration' interface!
[ros2_control_node-7] Joint 'wrist_3_joint' has to have 'position', 'velocity', or 'acceleration' interface!
```

I have tracked this message to this section of generic_system.cpp : 

https://github.com/ros-controls/ros2_control/blob/73d5dcac6f7109abd0c18319caf3fac972dbd097/hardware_interface/src/mock_components/generic_system.cpp#L397-L404

This error seems to appear because as the interfaces contain the name of the joint, it assumes they are the interface of that joint. This is due to this check here, which only checks for the joint name to be in the interface:

https://github.com/ros-controls/ros2_control/blob/73d5dcac6f7109abd0c18319caf3fac972dbd097/hardware_interface/src/mock_components/generic_system.cpp#L344-L347

I would suggest checking for the joint name to be in the start of the interface name instead.

christophfroehlich, 2025-05-19:
Can you please add the version of ros2_control you are using?

InigoMoreno, 2025-05-19:
> Can you please add the version of ros2_control you are using?

I am using the jazzy binaries. Currently building from source to try and fix it locally and make a PR.

InigoMoreno, 2025-06-18:
Fixed by #2120