ros-controls/ros2_control #2262: Document joint limiting behavior

christophfroehlich, 2025-05-21:
## Background

With https://github.com/ros-controls/ros2_control/pull/2047 we added a mechanism to limit the joints movements from limits defined in the URDF or manually.  

- Understand the configuration
- choose a proper place to update the [documentation](https://control.ros.org/rolling/doc/ros2_control/doc/index.html)
- Add it

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

HabibZRK, 2025-05-22:
Hey @christophfroehlich , 

I was looking for an entry point to open source contributions and I saw a good opportunity here. Can I please be assigned this issue?

Kindest of regards and thanks for the awesome support!

HabibZRK, 2025-05-29:
Hi @christophfroehlich,

While reviewing the changes from #2047 as part of documenting the joint limiting behavior, I noticed that in commit `9c4cd49` (https://github.com/ros-controls/ros2_control/commit/9c4cd4968c89df4786e40c1f31bd3acc43bf11dd), the joints `joint3`, `tool_joint`, and their corresponding links were added under the `urdf_head_continuous_missing_limits` test string (see diff: https://github.com/ros-controls/ros2_control/pull/2047/files#diff-e557193f04da56e8475ba7dfd70a1bf266107bf4447deb3facd017fe29532ea9). However, since `joint3` is a *revolute* joint and *has limits defined*, I was wondering if it might be more appropriate to place it under `urdf_head_revolute_with_limits` instead?

I just wanted to confirm:  
- Is this placement intentional for a specific testing purpose that I may have overlooked,  
or  
- Would you prefer that I open a separate issue/PR to move these elements to the more appropriate test case?

Additionally, since this is a documentation issue related to the `enforce_command_limits` parameter—which has differing default values between Rolling (`true`) and Jazzy & earlier (`false`)—would it be best practice to create two separate PRs? One targeting Rolling (to be backported to Kilted), and another for Jazzy and earlier distros?

Thanks very much for your patience! 😊

christophfroehlich, 2025-05-30:
You don't need to document the enforce_command_limits parameter, because this is already parsed and added here https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#parameters

@saikishor could you have a look please on the proposal?