ros-controls/ros2_control #2263: Ros2 control crashes when using with ur5 cb3.1

techtasie, 2025-05-21:
**Describe the bug**
A clear and concise description of what the bug is.
If launching the ur_robot_driver for a ur5 cb3.1 ros2 control crashes on me.
**To Reproduce**
Steps to reproduce the behavior:
1. Install ['Universal Robots ros2 driver'](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver)
2. Execute: `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=192.168.0.215`
3. Instantly Crashes

**Expected behavior**
A clear and concise description of what you expected to happen.
It should connect to the robot.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Fedora 42
 - Version Jazzy
- Anything that may be unusual about your environment: I source build all of ros2 jazzy on fedora with RelwithDebug

**Additional context**
Add any other context about the problem here, especially include any modifications to ros2_control that relate to this issue.
I attached the coredump. 

```ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=192.168.0.215
[INFO] [launch]: All log files can be found below /home/timwendt/.ros/log/2025-05-21-14-41-23-108720-god-352795
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [352806]
[INFO] [controller_stopper_node-4]: process started with pid [352808]
[INFO] [ros2_control_node-1]: process started with pid [352805]
[INFO] [robot_state_helper-3]: process started with pid [352807]
[INFO] [urscript_interface-5]: process started with pid [352809]
[INFO] [robot_state_publisher-6]: process started with pid [352810]
[INFO] [rviz2-7]: process started with pid [352811]
[INFO] [spawner-8]: process started with pid [352812]
[INFO] [spawner-9]: process started with pid [352813]
[dashboard_client-2] [INFO] [1747831283.431640718] [UR_Client_Library:]: Starting primary client pipeline
[controller_stopper_node-4] [INFO] [1747831283.435242980] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_helper-3] [INFO] [1747831283.435786763] [UR_Client_Library:]: Starting primary client pipeline
[robot_state_publisher-6] [INFO] [1747831283.436985934] [robot_state_publisher]: Robot initialized
[ros2_control_node-1] [INFO] [1747831283.499554175] [controller_manager]: Using Steady (Monotonic) clock for triggering controller manager cycles.
[ros2_control_node-1] [INFO] [1747831283.504956381] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-1] [INFO] [1747831283.514206622] [controller_manager]: update rate is 125 Hz
[ros2_control_node-1] [INFO] [1747831283.514283796] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1747831283.514420335] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-1] [INFO] [1747831283.515260943] [controller_manager]: Received robot description from topic.
[ros2_control_node-1] text not specified in the tf_prefix tag
[ros2_control_node-1] [INFO] [1747831283.523699071] [controller_manager]: Loading hardware 'ur5'
[ros2_control_node-1] [INFO] [1747831283.534613903] [controller_manager]: Loaded hardware 'ur5' from plugin 'ur_robot_driver/URPositionHardwareInterface'
[ros2_control_node-1] [INFO] [1747831283.535017585] [controller_manager]: Initialize hardware 'ur5'
[ros2_control_node-1] [INFO] [1747831283.535890249] [controller_manager]: Successful initialization of hardware 'ur5'
[ros2_control_node-1] [INFO] [1747831283.540315309] [resource_manager]: 'configure' hardware 'ur5'
[ros2_control_node-1] [INFO] [1747831283.540358052] [URPositionHardwareInterface]: Starting ...please wait...
[ros2_control_node-1] [INFO] [1747831283.540371112] [URPositionHardwareInterface]: Initializing driver...
[ros2_control_node-1] [WARN] [1747831283.541278208] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/real_time.html for details.
[rviz2-7] [INFO] [1747831283.709811626] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1747831283.709889987] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-7] [INFO] [1747831283.724156563] [rviz2]: Stereo is NOT SUPPORTED
[spawner-9] [INFO] [1747831283.742094264] [spawner_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
[dashboard_client-2] [INFO] [1747831283.884061884] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2]
[ros2_control_node-1] [INFO] [1747831283.941633974] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ros2_control_node-1] [INFO] [1747831283.942195566] [UR_Client_Library:]: Setting up RTDE communication with frequency 125.000000
[ros2_control_node-1] /usr/include/c++/15/bits/stl_vector.h:1263: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = double; _Alloc = std::allocator<double>; reference = double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[ros2_control_node-1] Stack trace (most recent call last) in thread 352941:
[ros2_control_node-1] #16   Object "/usr/lib64/ld-linux-x86-64.so.2", at 0xffffffffffffffff, in
[ros2_control_node-1] #15   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 78, in __GI___clone3 [0x7f5a1bd01ceb]
[ros2_control_node-1] #14   Source "/usr/src/debug/glibc-2.41-5.fc42.x86_64/nptl/pthread_create.c", line 448, in start_thread [0x7f5a1bc7f1d3]
[ros2_control_node-1] #13   Source "../../../../../libstdc++-v3/src/c++11/thread.cc", line 104, in execute_native_thread_routine [0x7f5a1be4dfe3]
[ros2_control_node-1] #12   Source "/usr/include/c++/15/bits/std_thread.h", line 255, in _M_run [0x40335b]
[... 60 lines cut ...]
[ros2_control_node-1]     |   110:       // Should never allocate memory because its capacity is able to hold all
[ros2_control_node-1]     |   111:       // variables
[ros2_control_node-1]     | > 112:       last_values.values[i] = references_[i].getValue();
[ros2_control_node-1]     |   113:     }
[ros2_control_node-1]     |   114:     last_values.values.resize(ref_size);
[ros2_control_node-1]     | Source "/usr/include/c++/15/bits/stl_vector.h", line 1261, in operator[]
[ros2_control_node-1]     |  1259:       _GLIBCXX_NODISCARD _GLIBCXX20_CONSTEXPR
[ros2_control_node-1]     |  1260:       reference
[ros2_control_node-1]     | >1261:       operator[](size_type __n) _GLIBCXX_NOEXCEPT
[ros2_control_node-1]     |  1262:       {
[ros2_control_node-1]     |  1263: 	__glibcxx_requires_subscript(__n);
[ros2_control_node-1]       Source "/usr/include/c++/15/bits/stl_vector.h", line 1263, in doUpdate [0x7f5a1aef0fe9]
[ros2_control_node-1]        1260:       reference
[ros2_control_node-1]        1261:       operator[](size_type __n) _GLIBCXX_NOEXCEPT
[ros2_control_node-1]        1262:       {
[ros2_control_node-1]       >1263: 	__glibcxx_requires_subscript(__n);
[ros2_control_node-1]        1264: 	return *(this->_M_impl._M_start + __n);
[ros2_control_node-1]        1265:       }
[ros2_control_node-1] #3    Source "../../../../../libstdc++-v3/src/c++11/assert_fail.cc", line 41, in __glibcxx_assert_fail [0x7f5a1be0b085]
[ros2_control_node-1] #2    Source "/usr/src/debug/glibc-2.41-5.fc42.x86_64/stdlib/abort.c", line 73, in abort [0x7f5a1bc0f6cf]
[ros2_control_node-1] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f5a1bc27afd]
[ros2_control_node-1] #0    Source "/usr/src/debug/glibc-2.41-5.fc42.x86_64/nptl/pthread_kill.c", line 44, in __pthread_kill_implementation [0x7f5a1bc8111c]
[ros2_control_node-1] Aborted (Signal sent by tkill() 352805 1000)
[ERROR] [ros2_control_node-1]: process has died [pid 352805, exit code -6, cmd '/home/timwendt/ros2/fix_ws/install/controller_manager/lib/controller_manager/ros2_control_node --ros-args --params-file /usr/lib64/ros2-jazzy/share/ur_robot_driver/config/ur5_update_rate.yaml --params-file /tmp/launch_params_cydda2z2'].
```

techtasie, 2025-05-21:
GitHub doesn't want me to upload the core dump here so i put it on sciebo: https://fh-aachen.sciebo.de/s/Zt6LNEDIIhHU1Rx

saikishor, 2025-05-21:
Hello @techtasie!

That's weird. Can you try building `pal_statistics` in Fedora? and which version are you building?. Can you also let us know how consistent is the crash?

https://github.com/pal-robotics/pal_statistics/blob/a740b599dbeaf8e91c9e9c87a0040fe36782a8f5/pal_statistics/src/registration_list.cpp#L108-L113

If this is very consistently happening, then it would be great if you can see what are the sizes before the crash.

Thank you

techtasie, 2025-05-21:
@saikishor, I tried building all 4.x versions of ros2_control but generated this coredump with 5.0.0. It happens every time.

Building pal_statistics in Fedora works just fine, but does not change anything regarding the crash. I should note that I disabled build tests.

techtasie, 2025-05-21:
In line 112 of `registration_list.cpp`:

```
(gdb) list .
107	    last_values.names = ids_;
108	    size_t ref_size = references_.size();
109	    for (size_t i = 0; i < ref_size; ++i) {
110	      // Should never allocate memory because its capacity is able to hold all
111	      // variables
112	      last_values.values[i] = references_[i].getValue();
113	    }
114	    last_values.values.resize(ref_size);
115	  } else {
116	    last_values.names.clear();
(gdb) p ref_size
$2 = 319
(gdb) p i
$3 = 4
```

saikishor, 2025-05-21:
Hello @techtasie,

It is weird that you have a length of 4. Can you add a condition like this before `all_enabled_`  ? `all_enabled_ = all_enabled_ && (last_values.values.size() == references_.size()); `, let's see if this fixes the issue and if it fixes, please share the output of [the introspection topic](https://control.ros.org/rolling/doc/ros2_control/doc/introspection.html#introspection-of-the-ros2-control-setup)

techtasie, 2025-05-21:
@saikishor, it works. Some ur components are now crashing, but I can figure that one out....

The changes I did:
```
diff --git a/pal_statistics/src/registration_list.cpp b/pal_statistics/src/registration_list.cpp
index 0878472..80207ff 100644
--- a/pal_statistics/src/registration_list.cpp
+++ b/pal_statistics/src/registration_list.cpp
@@ -103,6 +103,7 @@ void RegistrationList::doUpdate()
   assert(last_values.values.capacity() >= ids_.size());

   // This is for optimization, majority of the time everything is enabled and this runs 40% faster
+  all_enabled_ = all_enabled_ && (last_values.values.size() == references_.size());
   if (all_enabled_) {
     last_values.names = ids_;
     size_t ref_size = references_.size();
(END)
```

I attached the output of ros2 topic echo /controller_manager/introspection_data/full --once --full > 
[instrospection_data_full.txt](https://github.com/user-attachments/files/20368942/instrospection_data_full.txt)

saikishor, 2025-05-21:
I'm glad it worked for you. I'll see what i can do about this one.

techtasie, 2025-05-21:
@saikishor, if you need help evaluating something, just let me know.

techtasie, 2025-05-21:
Thanks for the fast help 👍

closed 2025-05-21 by techtasie