ros-controls/ros2_control #2274: Improve interface filtering function

bmagyar, 2025-05-24:
## Background

The `get_hardware_related_interfaces()` function in `hardware_interface/src/resource_manager.cpp` could use some improvements. 

The function itself could probably be renamed to convey accurately what it does. The choices of function name and argument names are all very task-specific while at it's core, the method is a simple intersection operation.
For more, read this comment in context: https://github.com/ros-controls/ros2_control/pull/2120#discussion_r2105772935

Memory management and algorithm runtime *may* be improved by pre-allocating the memory for the output vector based on the inputs. Verify which input vector is larger and re-organize the loop if needed. Also do the allocation based on the *smaller* vector.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ros2_control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

Amronos, 2025-05-25:
I would like to work on this, can this be assigned to me?

saikishor, 2025-05-25:
> I would like to work on this, can this be assigned to me?

@Amronos this issue is meant for the first time contributors.

Eldgar, 2025-05-26:
I am interested in doing this.

Eldgar, 2025-05-30:
I have created a pull request for this improvement.

A couple of notes on how I tested the change:

- Built the workspace locally with my ROS_DISTRO set to Jazzy, so I could compile the packages and run `colcon test` for `hardware_interface`. All tests passed.
- After confirming the tests, I reapplied the one-file patch on the `master` development branch (per the contribution guidelines) and opened this PR.

please let me know if any additional checks are needed or if there is a better way to do this.