ros-controls/ros2_control #2301: High TF Latency (~100ms) with ros2_control, diff_drive_controller, and CANopen Hardware Interface

alex-roba, 2025-05-29:
We are using ROS 2 Humble and observing a significant TF latency of approximately 100-102ms when using ros2_control with diff_drive_controller and a custom HardwareInterface that communicates with motor drives via CANopen.

This latency manifests visually in RViz: when the robot rotates, the Lidar scan (viewed in the odom frame) initially rotates slightly in the opposite direction before appearing stationary, and then "jumps" forward when rotation stops. This indicates that the TF transform is lagging behind the Lidar data by about 100ms.

We have benchmarked our HardwareInterface::read() method, and it consistently executes in approximately 2ms. Furthermore, tools like ros2 topic hz /odom confirm that the odom -> base_link transform is being published at a stable 100Hz, matching our configured update_rate.

Despite the fast hardware read and the stable publishing frequency, the ~100ms latency persists. This leads us to hypothesize that ros2_control (or the diff_drive_controller within it) might be publishing TF transforms using state data read from a previous cycle, but stamping these transforms with the current system time. This mismatch between the age of the data and the timestamp would precisely explain the observed ~100ms delay and the resulting visual effects in RViz.

Crucially, when we bypass ros2_control and run a standalone ROS 2 node (using the exact same CANopen communication and motor control code) to read sensors and publish odometry/TF directly, the TF latency is significantly reduced, and the "opposite rotation" effect in RViz is virtually eliminated. This strongly suggests the ~100ms latency is not inherent in our hardware communication but is introduced during the ros2_control update and publishing process.

christophfroehlich, 2025-05-29:
100ms would mean 10 update cycles delay? someone needs to debug this :/ 
fyi: from jazzy on we have diagnostics also for hardware components, which allows you to analyze the execution time of your read method for example. You can always compile the rolling version on humble, see "Development Version" instructions in the docs.

saikishor, 2025-06-23:
> @christophfroehlich I'm encountering the same issue described as Problem 3 in this [blog post](https://articulatedrobotics.xyz/tutorials/mobile-robot/applications/ros2_control-extra/). However, the difference is that while the author experienced it in simulation, I'm facing it on a real-world robot and opposite effect. The author has also provided a GitHub repository to help reproduce the issue.

@alex-roba Could you please check with the Rolling version and let us know if the issue happens?. You can compile the Rolling version of ros2_control using the instructions here : https://control.ros.org/humble/doc/getting_started/getting_started.html#building-from-source under the tab `Development version`. 

If you can let us know this, we can see what we can do to fix the issue