ros-controls/ros2_control #2309: `joint_state_broadcaster` fails to load with "Couldn't parse trailing --params-file flag. No file path provided." on ROS 2 Jazzy

lhbmv, 2025-06-03:
## Description

**Environment:**
- ROS 2 Distribution: Jazzy
- OS: Ubuntu 24.04 (noble) / [your arch, e.g. arm64]
- ros2_control: [e.g. 4.25.0-1noble]
- joint_state_broadcaster: [e.g. 4.25.0-1noble]
- Installed via: apt
- Workspace build: `colcon build --symlink-install`

**Problem:**

When launching a minimal ros2_control setup with only `joint_state_broadcaster` in the controller YAML, the controller fails to load with the following error:

```
[controller_manager]: Controller 'joint_state_broadcaster' node arguments: --ros-args --params-file 
[controller_manager]: Caught exception of type : N6rclcpp10exceptions22RCLInvalidROSArgsErrorE while initializing controller 'joint_state_broadcaster': failed to parse arguments: Couldn't parse trailing --params-file flag. No file path provided., at ./src/rcl/arguments.c:419
```

**controllers.yaml:**
```yaml
controller_manager:
  ros__parameters:
    update_rate: 50
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster
```

**launch file (excerpt):**
```python
robot_controllers = PathJoinSubstitution([
    FindPackageShare("planar_robot_description"),
    "config",
    "controllers.yaml"
])
control_node = Node(
    package="controller_manager",
    executable="ros2_control_node",
    parameters=[robot_controllers],
    output="both"
)
joint_state_broadcaster_spawner = Node(
    package="controller_manager",
    executable="spawner",
    arguments=["joint_state_broadcaster"]
)
```

**What I tried:**
- Checked YAML formatting, encoding (UTF-8, no BOM), and indentation (spaces only).
- Cleaned and rebuilt workspace (`rm -rf build/ install/ log/`).
- Used both PathJoinSubstitution and absolute path for the YAML file.
- Confirmed the controller type is available via `ros2 control list_controller_types`.
- Confirmed hardware and interfaces are available.
- The error persists regardless of these changes.

**Expected behavior:**
`joint_state_broadcaster` should load successfully as in previous ROS 2 distributions (e.g. Humble).

**Actual behavior:**
Controller fails to load due to a missing params file argument, even though the YAML is present and correct.

---

## Steps To Reproduce

1. Use the minimal YAML and launch file above in a clean ROS 2 Jazzy workspace.
2. Build and source the workspace.
3. Launch the file:  
   `ros2 launch planar_robot_description control_only.launch.py`
4. Observe the error in the controller_manager logs.

---

## Additional Info

- This issue does **not** occur on ROS 2 Humble or Rolling with the same setup.
- Other controllers (e.g. joint_trajectory_controller) exhibit similar issues.
- This blocks basic usage of ros2_control on Jazzy.

---

**Thank you for your help!**

saikishor, 2025-06-03:
Hello @lhbmv !

It could be that the 4.25.0 of Jazzy had that bug. Could you please update the package to the recent version and try it out?

Thank you

saikishor, 2025-07-07:
Hello @lhbmv,

Could you please confirm that this is solved?.

Thank you

GogiPuttar, 2025-08-20:
I have the same issue on 4.32.0:

```
ros2_control                              : latest=4.35.0, local=4.32.0
```

Here's my command:
```
ros2 run controller_manager spawner joint_state_broadcaster -c /controller_manager
[FATAL] [1755715708.370488269] [spawner_joint_state_broadcaster]: Failed loading controller joint_state_broadcaster
[ros2run]: Process exited with failure 1
```

Here's the gazebo side error message:
```
[INFO] [1755715708.347607273] [controller_manager]: Loading controller : 'joint_state_broadcaster' of type 'joint_state_broadcaster/JointStateBroadcaster'
[INFO] [1755715708.347672282] [controller_manager]: Loading controller 'joint_state_broadcaster'
[INFO] [1755715708.353323347] [controller_manager]: Controller 'joint_state_broadcaster' node arguments: --ros-args --params-file -p use_sim_time:=true --param use_sim_time:=true 
[ERROR] [1755715708.353618483] [controller_manager]: Caught exception of type : N6rclcpp10exceptions22RCLInvalidROSArgsErrorE while initializing controller 'joint_state_broadcaster': failed to parse arguments: Couldn't parse params file: '--params-file -p'. Error: Error opening YAML file, at ./src/parser.c:271, at ./src/rcl/arguments.c:415
```

And here's the controller config yaml:
```
# Tells controller_manager its update rate and which controllers exist (their types)
controller_manager:
  ros__parameters:
    update_rate: 400        # Hz; choose what you want CM to tick at
    use_sim_time: true

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

# Per-controller parameter namespaces (names must match the controllers above)
joint_state_broadcaster:
  ros__parameters: {}

joint_trajectory_controller:
  ros__parameters:
    joints: [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6, joint_7]
    command_interfaces: [position]         # what your hardware exposes for commands
    state_interfaces:   [position, velocity]  # what your hardware exposes as state
```

christophfroehlich, 2025-08-21:
what happens if you remove the empty joint_state_broadcaster section?

GogiPuttar, 2025-08-21:
Thanks for the response!

So with the following config file:
```
# Tells controller_manager its update rate and which controllers exist (their types)
controller_manager:
  ros__parameters:
    update_rate: 400        # Hz; choose what you want CM to tick at
    # use_sim_time: true

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

joint_trajectory_controller:
  ros__parameters:
    joints: [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6, joint_7]
    command_interfaces: [position]         # what your hardware exposes for commands
    state_interfaces:   [position, velocity]  # what your hardware exposes as state
```

I still get:
```
[INFO] [1755798669.532924622] [controller_manager]: Loading controller : 'joint_state_broadcaster' of type 'joint_state_broadcaster/JointStateBroadcaster'
[INFO] [1755798669.533035293] [controller_manager]: Loading controller 'joint_state_broadcaster'
[INFO] [1755798669.541626227] [controller_manager]: Controller 'joint_state_broadcaster' node arguments: --ros-args --params-file -p use_sim_time:=true --param use_sim_time:=true 
[ERROR] [1755798669.542135236] [controller_manager]: Caught exception of type : N6rclcpp10exceptions22RCLInvalidROSArgsErrorE while initializing controller 'joint_state_broadcaster': failed to parse arguments: Couldn't parse params file: '--params-file -p'. Error: Error opening YAML file, at ./src/parser.c:271, at ./src/rcl/arguments.c:415
```

GogiPuttar, 2025-08-21:
Absolutely, just give me 15-20 minutes!

GogiPuttar, 2025-08-21:
Here it is: https://github.com/GogiPuttar/ros2_control_bug_example/tree/main

I've tried to document everything properly in the README but please let me know if there are any issues with bringing this up. Thanks!

saikishor, 2025-08-22:
@GogiPuttar I've issues locally to run the commands

```
# gz sim -v4 empty.sdf
[Msg] Gazebo Sim GUI    v8.9.0
[Msg] Copied installed config [/opt/ros/jazzy/opt/gz_sim_vendor/share/gz/gz-sim8/gui/gui.config] to default config [/root/.gz/sim/8/gui.config].
[Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:275] Waiting for subscribers to [/gazebo/starting_world]...
[Msg] Received world [empty.sdf] from the GUI.
[Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].
[Msg] Gazebo Sim Server v8.9.0
[Msg] Loading SDF world file[/opt/ros/jazzy/opt/gz_sim_vendor/share/gz/gz-sim8/worlds/empty.sdf].
[Dbg] [Application.cc:96] Initializing application.
[Dbg] [Application.cc:170] Qt using OpenGL graphics interface
[GUI] [Dbg] [Application.cc:657] Create main window
[GUI] [Wrn] [Application.cc:908] [QT] QGLXContext: Failed to create dummy context
[GUI] [Dbg] [Application.cc:902] [QT] failed to acquire GL context to resolve capabilities, using defaults..
[GUI] [Dbg] [PathManager.cc:68] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Dbg] [Gui.cc:355] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:883] Loaded [gz::physics::dartsim::Plugin] from library [/opt/ros/jazzy/opt/gz_physics_vendor/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::Physics] for entity [1]
[Msg] Create service on [/world/empty/create]
[Msg] Remove service on [/world/empty/remove]
[Msg] Pose service on [/world/empty/set_pose]
[Msg] Pose service on [/world/empty/set_pose_vector]
[Msg] Light configuration service on [/world/empty/light_config]
[Msg] Physics service on [/world/empty/set_physics]
[Msg] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
[Msg] Enable collision service on [/world/empty/enable_collision]
[Msg] Disable collision service on [/world/empty/disable_collision]
[Msg] Material service on [/world/empty/visual_config]
[Msg] Material service on [/world/empty/wheel_slip]
[Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::Contact] for entity [1]
[Msg] Loaded level [default]
[Msg] Serving world controls on [/world/empty/control], [/world/empty/control/state] and [/world/empty/playback/control]
[Msg] Serving GUI information on [/world/empty/gui/info]
[Msg] World [empty] initialized with [1ms] physics profile.
[Msg] Serving world SDF generation service on [/world/empty/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[GUI] [Dbg] [PathManager.cc:57] Received resource paths.
[GUI] [Dbg] [Gui.cc:413] Requesting GUI from [/world/empty/gui/info]...
[GUI] [Dbg] [GuiRunner.cc:149] Requesting initial state from [/world/empty/state]...
[GUI] [Msg] Loading config [/root/.gz/sim/8/gui.config]
[GUI] [Dbg] [Application.cc:528] Loading plugin [MinimalScene]
[GUI] [Dbg] [MinimalScene.cc:802] Creating gz-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:802] Creating gz-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:986] Creating render thread interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:802] Creating gz-rendering interface for OpenGL
[GUI] [Dbg] [MinimalScene.cc:986] Creating render thread interface for OpenGL
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [MinimalScene] from path [/opt/ros/jazzy/opt/gz_gui_vendor/lib/gz-gui-8/plugins/libMinimalScene.so]
[GUI] [Dbg] [Application.cc:528] Loading plugin [EntityContextMenuPlugin]
[GUI] [Msg] Currently tracking topic on [/gui/currently_tracked]
[GUI] [Msg] Added plugin [Entity Context Menu] to main window
[GUI] [Msg] Loaded plugin [EntityContextMenuPlugin] from path [/opt/ros/jazzy/opt/gz_sim_vendor/lib/gz-sim-8/plugins/gui/libEntityContextMenuPlugin.so]
[GUI] [Dbg] [Application.cc:528] Loading plugin [GzSceneManager]
[GUI] [Msg] Added plugin [Scene Manager] to main window
[GUI] [Msg] Loaded plugin [GzSceneManager] from path [/opt/ros/jazzy/opt/gz_sim_vendor/lib/gz-sim-8/plugins/gui/libGzSceneManager.so]
[GUI] [Dbg] [Application.cc:528] Loading plugin [InteractiveViewControl]
[GUI] [Msg] Camera view controller topic advertised on [/gui/camera/view_control]
[GUI] [Msg] Camera reference visual topic advertised on [/gui/camera/view_control/reference_visual]
[... 78 lines cut ...]
#17   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7592cb1e1a7a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#16   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7592cb240278, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#15   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7592c8be3a62, in g_main_context_iteration
#14   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7592c8c43736, in 
[... cut ...]
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7592d19988fe, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7592d19b527d, in gsignal
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7592d1a0eb2c, in pthread_kill
Aborted (Signal sent by tkill() 155 0)
[Dbg] [SignalHandler.cc:142] Received signal[2].
[Dbg] [ServerPrivate.cc:122] Server received signal[2]
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic

```

GogiPuttar, 2025-08-22:
Okay, are you able to run just `gz sim` and `rviz2`? I'm trying to figure out if this an Intel GPU acceleration error, or just a display permissions error. 
What does `glxinfo | grep "OpenGL renderer"` say?