ros-controls/ros2_control #2362: Question regarding distributed control system with ros2_control

BartlomiejK2, 2025-07-02:
Hi! Me and my team are developing quadruped robot that consists of three computers:

1. Raspberry Pi - communicates with all motor controllers, onboard IMU and contact sensors
2. Intel NUC - Model Predictive Control algorithms
3. Jetson Nano - communicates with depth cameras, RL  algorithms

Because Raspberry Pi talks with hardware via `ros2_control`, I have to run `controller_manager` on it.

Is there a way to run MPC and RL algorithms on different computers as `ros2_control` controllers that talks with `controller_manager` on Raspberry Pi directly via its `hardware_interfaces`? 
Or the only solution is to run simple controller and broadcasters on Raspberry Pi and run other  algorithms as ROS2 nodes on different computers that uses these simple controller and broadcasters to get `hardware_interfaces`?

Juliaj, 2025-07-08:
@BartlomiejK2, regarding your question to run MPC and RL algorithms on different computers as ros2_control controllers, could you provide more context? For example, what would be the expected behavior of these controllers?

christophfroehlich, 2025-07-08:
There was a ROSCon talk by @mamueluth and @destogl  about distributed ros2_control, but only in German unfortunately https://www.roscon2023.de/presentations/S2_P3___ros2_control_ist_aus_meinem_Rechner_ausgebrochen.pdf

I'm not sure if they open sourced their changes.

BartlomiejK2, 2025-07-10:
@Juliaj our controller will include:

Multilayer perceptron (trained using RL in simulation before running on real hardware):
 - Get current robot feedback 
 - Generate and send parameters to non-linear MPC

Non-linear MPC:
- Get current robot feedback
- Get current parameters generated by MLP
- Generate and send new optimal joint commands

These joint commands will be send to hardware via hardware interfaces that are running on Raspberry Pi ([pi3hat_hardware_interface](https://github.com/BartlomiejK2/pi3hat_hardware_interface)).

@christophfroehlich thank you for the link! Unfortunately I could not find any repository on their github group :(

Juliaj, 2025-07-10:
@BartlomiejK2, your use case appears to differ from the one demonstrated in the ROSCon 2023 presentation. Since your MLP and MPC algorithms are purely computational and don't involve direct hardware interaction, IMHO, standard ROS 2 nodes might be a more suitable approach. Are there specific requirements that you'd like to achieve via a ros2_controller implementation?