ros-controls/ros2_control #2385: `controller_manager` spawner uses undefined logger style

SuperJappie08, 2025-07-11:
**Describe the bug**
When a controller spawner fails, when it has run out of attempts to acquire the lock.
This failing is fine however, it attempts to log with an undefined style from the `bcolors` class (`bcolors.ERROR`)

**To Reproduce**
Steps to reproduce the behavior:
1. Make the controller spawner timeout on waiting for a lock until the retry attempts runout.
2. Error occurs
4. See error (There are multiple controller spawners throwing multiple errors here)
```
[spawner-4] Traceback (most recent call last):
[spawner-4]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/spawner.py", line 199, in 
[spawner-4]     bcolors.ERROR + "Failed to acquire lock after multiple attempts." + bcolors.ENDC
[spawner-4]     ^^^^^^^^^^^^^
[spawner-4] AttributeError: type object 'bcolors' has no attribute 'ERROR'
[spawner-4]
[spawner-4] During handling of the above exception, another exception occurred:
[spawner-4]
[spawner-4] Traceback (most recent call last):
[spawner-4]   File "/opt/ros/jazzy/lib/controller_manager/spawner", line 33, in <module>
[spawner-4]     sys.exit(load_entry_point('controller-manager==4.32.0', 'console_scripts', 'spawner')())
[spawner-4]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[spawner-4]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/spawner.py", line 407, in main
[spawner-4]     node.destroy_node()
[spawner-4]     ^^^^
[spawner-4] UnboundLocalError: cannot access local variable 'node' where it is not associated with a value
[spawner-9] Traceback (most recent call last):
[spawner-9]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/spawner.py", line 199, in main
[spawner-9]     bcolors.ERROR + "Failed to acquire lock after multiple attempts." + bcolors.ENDC
[spawner-9]     ^^^^^^^^^^^^^
[spawner-9] AttributeError: type object 'bcolors' has no attribute 'ERROR'
[spawner-9]
[spawner-9] During handling of the above exception, another exception occurred:
[spawner-9]
[spawner-9] Traceback (most recent call last):
[spawner-9]   File "/opt/ros/jazzy/lib/controller_manager/spawner", line 33, in <module>
[spawner-9]     sys.exit(load_entry_point('controller-manager==4.32.0', 'console_scripts', 'spawner')())
[spawner-9]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[spawner-9]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/spawner.py", line 407, in main
[spawner-9]     node.destroy_node()
[spawner-9]     ^^^^
[spawner-9] UnboundLocalError: cannot access local variable 'node' where it is not associated with a value

```

**Expected behavior**
No uncaught exception.

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version Jazzy  (but also Kilted and Rolling)
- This only happens when spawning takes too long when using multiple spawners, however the error message should not crash. 

**Additional context**
https://github.com/ros-controls/ros2_control/blob/e905c07099e82fc950931812704901689b58018b/controller_manager/controller_manager/spawner.py#L191-L195

SuperJappie08, 2025-07-11:
I will make a pull request to fix this.

saikishor, 2025-07-11:
Hello!

The current jazzy version is not using node for the logging right?

SuperJappie08, 2025-07-11:
I have a PR ready for this

SuperJappie08, 2025-07-11:
> Hello!
> 
> The current jazzy version is not using node for the logging right?

I have applied my fix to the latest released Jazzy packages, and there is still the node error, which stems from it trying to clean up a Node, that has not been created.
That might be fixed in the latest commits, but I check that code now.

saikishor, 2025-07-11:
> I have applied my fix to the latest released Jazzy packages, and there is still the node error, which stems from it trying to clean up a Node, that has not been created.
> That might be fixed in the latest commits, but I check that code now.

Yes, please test the latest commits and let us know :)

SuperJappie08, 2025-07-11:
The extra commit fixes it

closed 2025-07-11 by christophfroehlich