ros-controls/ros2_control #2401: get_command helper does not use its temlate parameter

vvel-tracpilot, 2025-07-17:
**Describe the bug**
The `get_command` helper function always calls the double instantiation of the `Handle::get_optional` method, regardless of the template argument to `get_command`.

**To Reproduce**
Try to read a boolean command interface using `SystemInterface::get_command<bool>("...")`. This throws with:
> `runtime_error thrown during write of the component '<omitted>': std::optional<_Tp> hardware_interface::Handle::get_optional(std::shared_lock<std::shared_mutex>&) const [with T = double] failed. value_ptr_ is null.`

**Expected behavior**
This should return the boolean value obtained by calling `Handle::get_optonial<T> `with `T = bool`. 

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Version: Jazzy (deb packages)

**Additional context**
The related get_state function seems to work as intended, and, as far as I can tell, both were introduced by 5bc50b9. These functions are also not documented, and I found them through the examples in the examples in ros2_control_demos.

This seems to be an easy fix, and if this is the intended API for accessing state/command interfaces, I am happy to make a pull request.

vvel-tracpilot, 2025-08-26:
I have a fork that works, but I have not run any of the tests. The change is simple enough though. Feel free to use any of it, or I can open a pull request.

saikishor, 2025-08-26:
@ashwinsushil @vvel-tracpilot @JonasVervoort 

It is already fixed upstream right?

**Rolling:**
https://github.com/ros-controls/ros2_control/blob/4942abc760733e36a16e84eb882885f62dbe7150/hardware_interface/include/hardware_interface/hardware_component_interface.hpp#L585-L671

**Jazzy:**
https://github.com/ros-controls/ros2_control/blob/79eb6e04c73943c7739752807f6c688350220797/hardware_interface/include/hardware_interface/hardware_component_interface.hpp#L602-L688