ros-controls/ros2_control #2454: Controller Spawner launch extension

SuperJappie08, 2025-08-12:
**Is your feature request related to a problem? Please describe.**
I was attempting to do some dynamic controller configuration and remapping from a launch file (using the xml front-end).
During this I ran into parsing issues with the dynamically created `--controller-ros-args`, since `launch` attempts to split the arguments in an unescapable way (hardcoded it works fine, so no real issue in a more normal case).
Furthermore, it is currently not easily possible to set controller parameters that are defined only in a launch file. 

**Describe the solution you'd like**
A `launch` extension, like provided with `ros_gz_sim` and `ros_gz_bridge` in which special launch actions get exposed to spawn and unspawn hardware.
This could be part of the `controller_manager` package (similar to the Gazebo case) or a separate package (`launch_ros2_control`), similar to `launch_ros`. 

**Describe alternatives you've considered**
Writing a Python launch file, which is inconvenient since it is introducing a lot more more complexity to do something relatively simple.
While using a launch front-end (like `launch_xml` or `launch_yaml`) could also easily suffice.

**Additional context**
I would be willing to help realize this feature.

SuperJappie08, 2025-08-15:
I made a working version for Jazzy as a separate package called [launch_ros2_control](https://github.com/SuperJappie08/launch_ros2_control).
It currently only supports the most important features of the controller spawner.
(The main thing missing is remapping and parameters on the spawner itself).

I will add some more docs to it today.

SuperJappie08, 2025-10-01:
(nudge to keep alive, in case somebody is interested)

christophfroehlich, 2025-11-28:
@SuperJappie08 some of us had a look into your package and like it to add it to ros2_control repo. Could you open a PR please, so we can review it properly?
I think the examples could go to our demo repository?

mamueluth, 2025-11-29:
@SuperJappie08 very nice work! like @christophfroehlich mentioned i think it is a very nice addition and i want to give it a full review. 
Can you please create a package maybe ros2_control_launch and create a PR with it? Then we can review in detail and work towards merging this.

saikishor, 2025-11-29:
It looks great. I see that controller param files arg is missing right? Or maybe i overlooked it. 

https://github.com/ros-controls/ros2_control/blob/master/controller_manager%2Fcontroller_manager%2Flaunch_utils.py

Here you can find some existing utilities that we have to use with the launch file. Maybe this can help you when opening the PR.

SuperJappie08, 2025-11-29:
> It looks great. I see that controller param files arg is missing right? Or maybe i overlooked it. [...]

I can't remember exactly, but I think I might have missed the controller param files arg.
When I started on this, I planned/expected to be able to use `--controller-ros-args -p` to send individual parameters, but I couldn't make it work.
So I suspect it is a remnant of that original strategy.