ros-controls/ros2_control #2479: missing CMake targets on humble

christian-rauch, 2025-08-22:
**Describe the bug**
I am trying to replace `ament_target_dependencies` with `target_link_libraries`. In the suggested targets:
```CMake
target_link_libraries(my_robot_hardware PUBLIC
  hardware_interface::hardware_interface
  hardware_interface::mock_components
  rclcpp::rclcpp
)
```
none of the targets in the `hardware_interface::` namespace are available under `humble`.

Looking at the bloomed package, I can see that under `jazzy`, there is a `/opt/ros/jazzy/share/hardware_interface/cmake/export_hardware_interfaceExport.cmake` with the exported targets `hardware_interface::hardware_interface` and `hardware_interface::mock_components`. But there is no such equivalent file that exports these targets in the `humble` package.

**To Reproduce**
Steps to reproduce the behavior:
1. `find_package(hardware_interface REQUIRED)`
2. link a target with `target_link_libraries(my_hardware PUBLIC hardware_interface::hardware_interface hardware_interface::mock_components)`

You can also manually list the imported targets with:
```CMake
get_property(my_tgts DIRECTORY "${CMAKE_CURRENT_LIST_DIR}" PROPERTY IMPORTED_TARGETS)
foreach(t ${my_tgts})
    message(WARNING "  ${t}")
endforeach()
```
and see that those targets are not defined.

**Expected behavior**
The `hardware_interface` should export its CMake targets inside the `hardware_interface::` namespace.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version Humble

christian-rauch, 2025-08-24:
> Why do you need to change that on humble? `ament_target_dependencies` is still the [preferred](https://docs.ros.org/en/humble/How-To-Guides/Ament-CMake-Documentation.html#linking-to-dependencies) way to link against, and only changed from kilted on.

Using the same code for multiple distributions reduces the maintenance costs as I do not have to maintain multiple branches and backport between them. I was asked to apply this change (`ament_target_dependencies` -> `target_link_libraries`) on a couple of other repositories and there it worked without issues. So I guess this is an issue with the `hardware_interface` project.

christophfroehlich, 2025-08-24:
Gotcha. I'll trigger a backport of #926, let's see what happens and how many conflicts we have..

christian-rauch, 2025-09-01:
Fixed by https://github.com/ros-controls/ros2_control/pull/2480.