ros-controls/ros2_control #2484: Enable communication to an inactive hardware interface

Svastits, 2025-08-25:
In #2347 the controller switch mechanism was modified to disable activating a controller, if the hardware interface, whose command interface it would claim is not active.

This modification makes it impossible for inactive hardware interfaces to get any information from the ROS2 world (as creating a subscription is also not possible), therefore it is no longer possible to set certain runtime parameters of the hardware interface, necessary for starting the control loop.

This change breaks the [kroshu ](https://github.com/kroshu) driver, that implements hardware interfaces for all KUKA robot OS-s

**Example:** 
It is possible to set the cycle time of an industrial robot before the control loop is started. This configuration was until now possible using a custom controller, that changed a command interface (=member variable) of the hardware interface, and the `on_activate()` callback could take this into account. Now the `on_activate()` can only start the control loop with the deafult cycle time, as it is no longer possible, to transfer this information in runtime, before the hardware interface is activated.

**Proposal**
If it is not feasable to enable controller activation in such a case by default, I would suggest to make this configurable, so that some inactive command interfaces can be claimed. (Something similar is mentioned in #2318 in a different context: some controllers have to remain active after an error to make error recovery possible)

MarqRazz, 2025-08-25:
I apologize that this has broken your system but we did it to improve the safety of our framework and make better design patters for how it should be used. Adding the ability to write to an inactive command interface opens the door to all kind of safety concerns and undefined behavior that people can implement and makes it even more difficult to keep the behavior of ros2_control well defined. In this undefined mode which interfaces are allowed and which ones will make it crash or hurt someone?

We have a few options in place for what you describe.

All `hardware_interfaces` include a `hardware_interface::HardwareInfo info_` member variable that can store and retrieve configuration variables for the hardware. For example here is how to set the [ip_address](https://github.com/locusrobotics/ros2_kortex/blob/main/kortex_description/arms/gen3/7dof/urdf/kortex.ros2_control.xacro#L48) (and lots of other configuration values) for the robot to read in [on_init()](https://github.com/locusrobotics/ros2_kortex/blob/main/kortex_driver/src/hardware_interface.cpp#L97), `on_activate()` or any method in the derived hardware_interface.

You also still have the option to only activate your custom controller and write to the hardware to re-configure it as long as it's interface is active. After you have re-configured you can start the joint command interfaces and continue...

We do have the want/desire to make it possible for hardware to know which interfaces are active and allow each one to say if commands can be written to them but this feature has not been implemented yet. We welcome contributions if you would like to tackle this, but would recommend coming with a design doc if you'd like to tackle this large of a feature 😄

Svastits, 2025-08-25:
I agree that this is a good direction to limit misuse of `ros_control`, but without the ability to configure an exception for certain controllers, this change disables a few valid use cases.  #2318 was closed, although the second step mentioned in the description seems to be missing. Do you know if somebody is already working on this concept?
If not, do you have detailed requirements on how such a change should look like? We have limited knowledge of the inside of ros2_control and cannot really estimate the required effort

gavanderhoorn, 2025-08-25:
Is there any documentation on the backporting policy of `ros2_control`?

The PR referenced by @Svastits seems like it introduces a rather breaking change -- on the behavioural side only perhaps, but still. Should that have been backported to an already released LTS like Jazzy?

MarqRazz, 2025-08-25:
> Is there any documentation on the backporting policy of `ros2_control`?

We are [working](https://github.com/ros-controls/control.ros.org/pull/481) on an official stance there too.

> If not, do you have detailed requirements on how such a change should look like? We have limited knowledge of the inside of ros2_control and cannot really estimate the required effort

We have been having monthly meetings for quite some time about this exact feature and the distros it should be back ported to. Please feel free to join and discuss any use cases you might have.

gavanderhoorn, 2025-08-25:
@MarqRazz wrote:
> @gavanderhoorn wrote:
> > Is there any documentation on the backporting policy of `ros2_control`?
> 
> We are [working](https://github.com/ros-controls/control.ros.org/pull/481) on an official stance there too.

`ros2_control` has a bit of a reputation -- as a project -- for regularly breaking downstream projects.

Am I correct in assuming that until that PR is merged this will continue to happen?

(note: I just happened to notice the KUKA driver breakage, I'm not involved in their development).

@saikishor wrote:
> Moreover, the fact that there is an active controller (or) to be able to activate a controller for an inactive hardware didn't make sense to us.

"*didn't make sense*" doesn't sound like a very strong argument for introducing breaking changes like this.

> we have discussed about this in the PMC meetings

which seems appropriate for a change like this.

However, reviewing the [commits](https://github.com/ros-controls/ros2_control/pull/2347/commits) the rationale for making these changes doesn't appear to be captured in the messages. Is that an oversight perhaps?

Svastits, 2025-09-02:
@saikishor thanks for the fixes, eveything seems to be fine
With setting the new parameter to true, our driver can properly start and after deactivation the controllers are still active :)

closed 2025-09-27 by bmagyar