ros-controls/ros2_control #2492: Jazzy branch gives CMake error

dcconner, 2025-08-26:
**Describe the bug**
After cloning and changing to the Jazzy branch, I get the following error:

`Make Error at CMakeLists.txt:4 (find_package):
  By not providing "Findros2_control_cmake.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ros2_control_cmake", but CMake did not find one.

  Could not find a package configuration file provided by
  "ros2_control_cmake" with any of the following names:

    ros2_control_cmakeConfig.cmake
    ros2_control_cmake-config.cmake

  Add the installation prefix of "ros2_control_cmake" to CMAKE_PREFIX_PATH or
  set "ros2_control_cmake_DIR" to a directory containing one of the above
  files.  If "ros2_control_cmake" provides a separate development package or
  SDK, be sure it has been installed.
`

**Expected behavior**
The package should be standalone with default ROS 2 Jazzy install.

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version: Jazzy

dcconner, 2025-08-26:
Not CMake issue, a separate package that is needed .  `sudo apt install ros2-ros2-control-cmake`

closed 2025-08-26 by dcconner

christophfroehlich, 2025-08-27:
Yes, we use that from jazzy on. `rosdep` should also help here.