ros-controls/ros2_control #2592: Option for uncolored log output

christophfroehlich, 2025-09-30:
**Is your feature request related to a problem? Please describe.**
Our CI logs always look like
`[NON-XML-CHAR-0x1B][96mSetting controller param "type" to "imu_sensor_broadcaster/IMUSensorBroadcaster" for [NON-XML-CHAR-0x1B][1mimu14_sensor_broadcaster[NON-XML-CHAR-0x1B][0m`

**Describe the solution you'd like**
From https://docs.ros.org/en/humble/Concepts/Intermediate/About-Logging.html

> RCUTILS_COLORIZED_OUTPUT - Control whether colors are used when outputting messages. If unset, automatically determine based on the platform and whether the console is a TTY. If 0, force disable using colors for output. If 1, force enable using colors for output.

Let's use the same logic as (tbd: is there an equivalent of `isatty` for python?)

https://github.com/ros2/rcutils/blob/3bbdf935477c2ba999943085abd4fbfe8d2867ce/src/logging.c#L1400-L1406

and don't add these colors in spawners etc

https://github.com/ros-controls/ros2_control/blob/bb87986e7cba6bc01ef77ab480a964d8fa9189ec/controller_manager/controller_manager/controller_manager_services.py#L41-L51

saumanraaj, 2025-10-21:
Hi, I’d like to take this issue.
I plan to implement an environment- and TTY-aware color control for the Python log output (respecting RCUTILS_COLORIZED_OUTPUT, if needed d fallback to isatty() and an optional --no-color flag).
I’ll submit a draft PR once I have a working version—please let me know if there are any additional preferences before I start.

christophfroehlich, 2025-10-22:
Very welcome! Have a look at our [contribution guidelines](https://control.ros.org/rolling/doc/contributing/contributing.html), and feel free to submit a draft if you want to get early feedback.

saumanraaj, 2025-10-25:
Hey,
I’ve created a Pull Request addressing this issue: https://github.com/ros-controls/ros2_control/pull/2741

The change makes the bcolors class environment-aware (respecting RCUTILS_COLORIZED_OUTPUT), removes ANSI escape codes from all node.get_logger() calls, and maintains full backward compatibility with existing usage.

closed 2025-11-07 by christophfroehlich